gysmo38/mitsubishi2MQTT #4: No abre pagina control

mappolo, 2019-09-13:
Cuando intento entrar en la pagina de control no consigue encontrarla, y tarda mucho en acceder a las diferentes paginas.

gysmo38, 2019-09-13:
Please try in English

mappolo, 2019-09-13:
When I try to enter the control page, I can't find it, and it takes a long time to access the different pages.

gysmo38, 2019-09-16:
It should work with last update

closed 2019-09-16 by gysmo38

endeavour, 2019-12-22:
I have 2 controllers, 1 works perfectly fine, but the other one has the same issue as described above.
When I try to access the control page the esp8266 reboots itself and the page never loads. Both have the same hardware (adafruit feather huzzah), so not sure what's going on here.

endeavour, 2019-12-22:
Turns out that this happens if the connection to the heatpump isn't wired up and/or has a loose connection.