gysmo38/mitsubishi2MQTT #6: Stability problem and suggestion

dzungpv, 2019-09-16:
I have 4 Wemos D1 with 4 Aircons, 3 of them running default https://github.com/SwiCago/HeatPump/tree/master/examples/mitsubishi_heatpump_mqtt_esp8266_esp32 and hass custom component and working fine for months.
But i test 1 with your firmware, it hang after about a week, only turn off power of the device and turn back on it will running again.

So may be it is memory leak or not enough memory with your firmware, so i give some suggestion:
**-  Web control interface and logs/console should make optional, have option to enable it in the web settings page.**
**- Serial.Print/Serial.Println should use macro like Debug.out(string)f or easy disable**

gysmo38, 2019-09-16:
I am using ESP8266 ESP-01 without problem. Maybe a problem with Wemos D1.  I will try to code your suggestions. Your are not the first one to report me  stability problem with Wemos D1.

dzungpv, 2019-09-19:
@gysmo38 i try your latest code, but the HTML control not display, i check source still but white page on both Chrome and Firefox

dzungpv, 2019-09-22:
@gysmo38 I updated about the stability issue, like i said before i use default SwiCago 's example without any issue for month with Wemos D1, but with your code it still repeat the problem:
- About 4 days, mqtt command not respond again and temperature not update
- Web interface still access, mean the chip not hang, but it not respond with web control command, i turn it off but it not work, web interface update to previous state.
- It only work after i use "Reboot" button on the web interface

closed 2019-09-23 by gysmo38