gysmo38/mitsubishi2MQTT #11: Support for setRemoteTemperature

alekslyse, 2019-09-28:
Any fork or implementation of the method void HeatPump::setRemoteTemperature(float setting) so a custom temperature override can be pushed via mqtt? (since the heatpumps usually got a bad reading of temperature from the builtin sensor)

gysmo38, 2019-09-29:
I will look when I have time. If you want to make a PR, you are more than welcome :)

alekslyse, 2019-09-29:
Great. I did make a fork and implemented it, can push it when I got sorted out the esp behaviour issues.

Though, what boards is this tested on and what flash and spffs is suggested?

Tried to run it on a nodemcu with 2mb for code and 2 for spffs and it's a bit wonky. After spffs reset the captive portal show up without CSS, and buttons do nothing. 

Also don't forget about the max mqtt packet size that will fail the state update on longer topics.

closed 2019-10-29 by gysmo38