gysmo38/mitsubishi2MQTT #14: Stuck after MQTT settings

brusarp, 2019-11-17:
Hi,
I've succesfully programmed my D1 mini (maybe clone??) with your git.
After connection to AP and set the SSID and the PW it correctly reboot and connect to my wifi.
If I try to setup the MQTT settings it stuck to "Saving configuration and rebooting... Refresh in 10s..." and nothing happens (the D1 stay connected to wifi as I can ping it correctly).
Tell me if I can to help you to found the problem with logs or somewhat (tell me how to do)
(PS: I've also try the dzungpv git with same error...)
Thank you

Tenebruspt, 2019-11-21:
Hi

I'm having the same issue. How did you solve it?

I don't use username / password on my mqtt server so left those parameters empty. Was this the same as your problem?

Thanks

tbrasser, 2019-11-24:
Hmm, mine was working somewhat, but got stock after a time. So I did a reboot by saving the existing (working) MQTT settings, and now it doesn't come back online. 

Time for a power cycle :(

gysmo38, 2019-12-15:
Hello,
I am not sure blank login and password work.  Can you try to put login and password on mqtt to test if it work. 
I am also having connection problem, I need some time to reboot with the web interface to get it work on HA. I am working on it.

gysmo38, 2019-12-15:
I rewrite code for MQTT. It should work better. Now it reconenct automatically to MQTT server if there is a network failure. I also add a status page to monitor MQTT status. There is a code between brackets:

-4 : MQTT_CONNECTION_TIMEOUT - the server didn't respond within the keepalive time
-3 : MQTT_CONNECTION_LOST - the network connection was broken
-2 : MQTT_CONNECT_FAILED - the network connection failed
-1 : MQTT_DISCONNECTED - the client is disconnected cleanly
0 : MQTT_CONNECTED - the client is connected
1 : MQTT_CONNECT_BAD_PROTOCOL - the server doesn't support the requested version of MQTT
2 : MQTT_CONNECT_BAD_CLIENT_ID - the server rejected the client identifier
3 : MQTT_CONNECT_UNAVAILABLE - the server was unable to accept the connection
4 : MQTT_CONNECT_BAD_CREDENTIALS - the username/password were rejected
5 : MQTT_CONNECT_UNAUTHORIZED - the client was not authorized to connect

Hope it will work better. I update my five units. I will tested dor severals weeks :)

endeavour, 2019-12-22:
Make sure you set the MQTT port explicitly. If you leave the textbox empty it doesn't work.

gysmo38, 2019-12-29:
It should work with last commit.

closed 2019-12-29 by gysmo38

brusarp, 2019-12-30:
Thank you @gysmo38 for your test and solution... i'm sorry for my late response (family trouble....) but I'll check as soon as possible!