gysmo38/mitsubishi2MQTT #15: web /control is working but connection to HA publishes to mqtt /debug topic

tbrasser, 2019-12-01:
This was really easy to set-up, thanks a lot!

I'm still having a few issues though, I get the entity in Home Assistant, but in MQTT logs I can see that after an initial config and status message, it goes to sending debug msges only:

"failed to publish hp status change"

And also, the current temperature tpl seems to fallback to 26 degrees in home-assistant (as in the arduino sketch is defined), but in the webinterface on the /control page, it does show the correct current temperature.

Is something going wrong with setting up the json mqtt message?

gysmo38, 2019-12-15:
Yes I notice this also. I will check what happen.

gysmo38, 2019-12-18:
Can you try with the last commit. I hope it will work as expected.

tbrasser, 2019-12-18:
I'll try as soon as I have access!

endeavour, 2019-12-21:
Also seeing this issue (the 26c). Also HA keeps trying to put the heatpump into HEAT mode even when I click on cool?

endeavour, 2019-12-22:
Seems like you are publishing to the ```ha_state_topic``` in multiple places but only in ```hpStatusChanged``` is roomTemperature set... therefore when the topic is published to from any of the other places it breaks the HA UI

endeavour, 2019-12-22:
https://github.com/gysmo38/mitsubishi2MQTT/pull/16

gysmo38, 2019-12-22:
Ok I will fix this

tbrasser, 2019-12-23:
I have performed an OTA (but was connected via VPN so had to do it from commandline specifying IP Port (and password). It succeeded, now waiting for it to come online (hopefully it didn't overwrite/reset my wifi settings).

Will it automatically come back online or do I need to power-cycle it  (or worst case, reconfigure the network settings)?

Also, thanks a bunch for the fixes these past days!

gysmo38, 2019-12-23:
If you uploaded only sketch, it should come back online. If you upload all the flash, you will loose your configuration.

gysmo38, 2020-01-22:
no news I close .

closed 2020-01-22 by gysmo38