gysmo38/mitsubishi2MQTT #18: Issues - latency and so on the user interface through HA

jascdk, 2019-12-30:
Hi Gysmo.

Well - i just got your code flashed to an esp-01 module with an esp8285 mcu on it. I think i have done it right - the code runs and the webpages show correctly. 

BUT - i see some issues on running through HA. When i set / change settings, there is some kind of long latency and when i turn down the temperature it just bounces back on a much higher temperature . Any thoughts what might cause this?

Best

Jacob

gysmo38, 2020-01-02:
Yes I knwon about the latency between change on Unit and HA. I am working on it.
For the temperature, it go to 31°C ?

jascdk, 2020-01-02:
regarding temperature - yes it goes to max 31 degrees:)

gysmo38, 2020-01-17:
I have commit an update. You should have no more latency on HA interface :)

jascdk, 2020-01-18:
@gysmo38 thanks for your great work! i am going to upload it now - will get back to you with my experiences;)

gysmo38, 2020-01-22:
I will close this issue because I think it is resolved in last commits. Open a new one if needed.

closed 2020-01-22 by gysmo38