gysmo38/mitsubishi2MQTT #21: Test V4.5

alexmc1510, 2020-01-07:
Hello @gysmo38 , thanks a lot for your effort. I am testing your code in one of my units: MSZ-HR35 and it works almost perfect.
Below you can find what I have detected:
- Differences between current temperature. After a while (5min) Hassio has been updated with the correct value but it should be instantaneous.
[image]
- After each control (target modify, heating, cooling....) the thermostat component returns to the previous value and after some seconds show the order. It seems that takes time to refresh. I can make a movie if you want.

I will continue testing it but the job done is ENORMOUS.

Thanks a lot.

jascdk, 2020-01-08:
@alexmc1510 hi :) i am experiencing this exact same issue . The modes have a lot of delay between switching in home assistant . The temperature bounces back when adjusting it . Very strange . Not working smooth at all

austwhite, 2020-01-08:
As a note.  I had the same issue with the custom component as well so this may be a MQTT delay in Home Assistant

alexmc1510, 2020-01-08:
I confirm that control webpage works fine. Without delay. The problem comes when using HA and MQTT.

alexmc1510, 2020-01-09:
Would someone be so kind to have a look. Sorry but I am not an expert on programming so I have no idea where to start. I confirm that the other 2 HVAC have the same problem. Current temperature doesn't update.
Thanks

gysmo38, 2020-01-10:
Hello I will check. I have no time right now but I will check this weekend or next week.

alexmc1510, 2020-01-20:
Hello @gysmo38 , did you have the chance to check it? If you tell me more or less where could be the issue I can try to help you. 
Thanks,

gysmo38, 2020-01-20:
I have made an update, the delay problem should be ok now.

alexmc1510, 2020-01-20:
Hello @gysmo38 , thanks for the update. As soon as I returned home I checked it with the following results:
- The delay existing after click on buttons has dissapeared. The button changes inmediately, 2 seconds after, return to the previous status but it only "flash". I can record a movie if you need it.
- The current temperature of the room remains on 26º always. I have checked the code and it's the default value when data cannot be parsed to HA.

All the other things works great. I have noticed an abnormal current temperature of the room but this is another topic, maybe related with swicago lib or maybe my unit....but I don't know how could I check it...

Thanks in advance.

Regards,

austwhite, 2020-01-21:
Your current room temperature issue is nothing to do with the SwiCago library.  
The solution by Unixko and the original custom component have no issue with current room temperature.
To be honest, I haven't had the current room temperature issue with this solution either.  I'll have a deeper look to see if I can replicate it.
Just wanted to make sure you don't waste time looking at the library for the issue. 🙂

gysmo38, 2020-01-22:
@alexmc1510 
For the flash, I notice but I not find a solution yet... It is because when you send an order to the unit, it take time for the unit to refresh is state. The unit send us wrong information.

For room temperature, can you try to check mqtt message from unit? You can use mqtt exlporer. If your room temp stay at 26 degrees, it look like your unit not sening the room temperature.

alexmc1510, 2020-01-23:
Hello @gysmo38 , I confirm that I receive data and match webpage info. What I notice is that room temperature is far from the real. In this moment, I have a thermostat just below the unit and it has 19ºC but unit is showing 13ºC. Do you know where the problem could be located? I don't know where I could check real unit temperature just to verify that the temperature I am receiving is the real temperature the unit is reading or if the problem comes from the serial decoding.

Thanks,

2920, 2020-02-06:
Does anybody want to test this?

gysmo38, 2020-03-27:
Another topic exist

closed 2020-03-27 by gysmo38