gysmo38/mitsubishi2MQTT #22: No support for Fahrenheit, which is needed for locales who use imperial scale

shampeon, 2020-01-13:
Unlike an HA custom component, which can set the display scale to follow the unit system of HA, an autodiscovered MQTT hvac device can't be configured to use Fahrenheit to display and set the temperature. HA assumes that all devices follow the scale set if you set `unit_system` in your configuration.
```
homeassistant:
  unit_system: imperial
```

Since I live in the US, installing mistsubishi2MQTT displays the climate component in HA as if the values were Fahrenheit when they are actually Celsius.
[image]

The code assumes the use of metric scale, probably since the SwiCago library and the Mitsubishi units themselves use Celsius internally.

The sketch should have a configurable option for displaying and processing Fahrenheit values and converting them to and from Celsius when interacting with the SwiCago library.

shampeon, 2020-01-13:
Added https://github.com/gysmo38/mitsubishi2MQTT/pull/23 to fix this issue.

shampeon, 2020-01-14:
Fixed with #23.

shampeon, 2020-01-19:
Your latest commit (https://github.com/gysmo38/mitsubishi2MQTT/commit/6391f332991ab0dbc1afa11b935481d7e826546e) overwrote all the Fahrenheit code you merged in with #23.

gysmo38, 2020-01-20:
Ok sorry about that. I maybe made a mistake. I will put your code again.

gysmo38, 2020-01-20:
I push a new commit with your Fahrenheit code :) Hope it is ok now.

shampeon, 2020-01-20:
Ok, thanks.

shampeon, 2020-01-20:
Spoke too soon, you forgot the HA config templates that were altered to support F. Fixed in #24.

gysmo38, 2020-01-20:
Thank you for your PR.
I have the idea to add a Unit menu under Setup. In this page we will able to choose between metrecis and fahrenheit.

closed 2020-01-21 by shampeon