gysmo38/mitsubishi2MQTT #31: Still problems with temp bouncing back and forth in latest version

jascdk, 2020-02-17:
Hi. Just installed the latest version.

When im controlling the temperature it is very inconsistent. At my device I need it to go down to minimum 10 degrees. That I set up in the config file before flashing. 

I made a video, where it shows the thing im getting. Its funny it I always going back to 26 degrees when I go down to 10 degrees.

Why is that happening?

https://youtu.be/7mfNOR5UGmg

jmvaz, 2020-02-18:
Does your hvac support 10 degrees? From my knowledge Mitsubishi supports from 16 to 31 degrees. What you are doing is one overflow on the supported values. Fix your configuration.

jascdk, 2020-02-18:
Well it has a ISave function which lowers the temperature to 10 degrees. Maybe I should try to load the firmware once more leaving the temp between 16 and 31 degrees . Do the code does not support lowering it or could it be changed in the code you think ?

lesleypersyn, 2020-02-18:
please edit: https://github.com/gysmo38/mitsubishi2MQTT/blob/89bc404cc3f5b9e9a7acb8b18e871e78b03cb2fd/src/mitsubishi2mqtt/config.h#L57

jascdk, 2020-02-18:
> please edit:
> 
> https://github.com/gysmo38/mitsubishi2MQTT/blob/89bc404cc3f5b9e9a7acb8b18e871e78b03cb2fd/src/mitsubishi2mqtt/config.h#L57

to 10 or back to 16?  I tried to change it to 10, but that messed up everything

shampeon, 2020-02-20:
You're basically hitting the dummy default value for the temperature setting.
https://github.com/gysmo38/mitsubishi2MQTT/blob/master/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L965

I don't know exactly why, but the the template in HA defaults to 26 degrees C if the setting is undefined or below 16 degrees C. It could be a delay in getting the values set in MQTT.

jascdk, 2020-04-10:
Hi. I still getting this "error". My system (european) has the option to set it to an "ISave" function via the remote control. All it does is to make the system go below to 10 degrees for keeping as low as possible during wintertimes (good for the house and good for saving energy). The normal temperature range is from 16 to 31 degrees. But when use this code - and lowering the "float min_temp" to 10 instead of 16 it messes up the control of the heatpump by always knocking it up to 26 degrees via HA. Very annoying :( Is there a fix for that?

gysmo38, 2020-04-10:
I've got same function on mine. I will made test.

shampeon, 2020-04-10:
I think the fix is to set `temp_stat_tpl` to either the `min_temp` or `max_temp` if the temperature is set to something outside that range, and something right in the middle if it's undefined.

The Jinga template is something like this (in Fahrenheit):
```
{% if (value_json is defined and value_json.temperature is defined) %}
{% if (value_json.temperature|int > 50 and value_json.temperature < 90) %}
{{ value_json.temperature }}
{% elif (value_json.temperature < 50) %}
50
{% elif (value_json.temperature > 90) %}
90
{% endif %}
{% else %}
72
{% endif %}
```

In the code, replace 50 and 90 with `(String)getTemperature(min_temp, useFahrenheit)` and `(String)getTemperature(max_temp, useFahrenheit)`.

gysmo38, 2020-04-11:
With last commit, it should resolve 16° bug.
For isave is more complicate because Swicago library allow between 16 and 31° so 10° will not work.
I made some debug and  if I active isave or econo cool on the remote, it send packet no implemented. I ask for help on Swicago gitter.

gysmo38, 2020-04-28:
I have answer from Swicago on gitter. You can view it here https://gitter.im/Mitsubishi-Heat-Pump/Lobby#
In fact nobody find a way to see if isave or econo cool are activated by remote. When I put isave on hvac and 10degrees in heat mode, it answer heat mode and 16 degrees heat. I think it heat at 16 degrees and stop or start by check room temp. You can simulate this by an automatisation in HA.

gysmo38, 2020-06-28:
I think first problem resolv :)

closed 2020-06-28 by gysmo38