gysmo38/mitsubishi2MQTT #32: Control interface disappears after loading

shampeon, 2020-02-22:
After flashing and setup the control interface will stop loading. The web console shows an error in the response, and no HTML is in the source.

I think we might be hitting the upper size limit of String sent as a response. https://github.com/esp8266/Arduino/issues/3205

jascdk, 2020-02-24:
I have the same problem - the control interface is gone :/

Shadowpheonix, 2020-02-24:
I too am experiencing this issue.

shampeon, 2020-02-24:
I'm almost certain it's a problem with the size of the response we're sending, or the String buffer we're using. If you remove the form elements below Fan from html_page_control, the control panel appears. I removed everything and began adding back in form elements one by one. I can get to VANE_2, but adding the option for VANE_3 breaks the control panel.

chrdavis, 2020-02-25:
I have a PR out with a fix for this.  I send the header, control page, and footer as separate chunks which works.

gysmo38, 2020-02-25:
I tested the PR on my 5 unit, it look good. I close the issue.

closed 2020-02-25 by gysmo38