gysmo38/mitsubishi2MQTT #37: Interaction with Domoticz

spfmoby, 2020-03-19:
I thought that your esp8266 plugin was based enough on the Swicago MQTT code so this domoticz plugin would work: https://github.com/Masurov/Domoticz-MitsubishiHpMQTT-Plugin

But it seems that they use different "words" to exchange through MQTT.
For example the domoticz plugin sends commands like:
pac/pac/set {"vane": "SWING"}
pac/pac/set {"mode": "FAN"}

But your code sends commands/status like:
pac/pac/state {"roomTemperature":19,"temperature":19,"fan":"AUTO","vane":"1","action":"off","mode":"off"}
pac/pac/settings {"temperature":19,"fan":"AUTO","vane":"1","mode":"dry"}

Is there a way to make the two communicate? Have you developed a plugin for domoticz for your version of the code?

shampeon, 2020-03-22:
The Swicago code has nothing to do with MQTT. It's strictly a library for interfacing with the Mitsubishi head unit using the serial bus. It did include examples of reading and setting options to and from MQTT topics, but all the various MQTT interfaces (HASS, OpenHAB, Domoticz) occur on a layer above the Swicago code.

It's not trivial to just change the MQTT topics. It looks like the Masurov plugin only uses a single MQTT topic to set options. This project has a setting topic as well as a state topic. The HASS autodiscovery topic maps these so HASS can use them. In other words, the state and status topics aren't specific to HASS.

A better fix for this, IMO, is modifying the Masurov plugin to use the generic state and setting topics provided by this project.

gysmo38, 2020-04-11:
@spfmoby Do you try to contact Masurov to see if he can adapt his plugin to our mqtt topics?

spfmoby, 2020-04-12:
Yes... and he almost said what you said (but opposite): a better fix for this is that gysmo38 sticks to the original Swicago MQTT topics/states from the examples as he can't develop a version of his domoticz plugin for every project based on the Swicago library.

He forked your project https://github.com/Masurov/mitsubishi2MQTT to remove your MQTT implementations and put back the "original" ones. But that' not a very clever solution as he forked it for me so I can try, but he won't maintain it.

Maybe from what he has done in his fork you can take his code, and integrate a radio button in the parameters so we can switch between the two MQTT implementation?

gysmo38, 2020-04-12:
I think it is not a good idea because he break homeassistant integration.
We also start from the example but we made modification to work with homeassistant. 
I am agree with, a configuration option could be a solution. The problem, I am not using Domoticz anymore, so I can not implement it. If he can do it and send a PR, that could be great.
I will check the fork if I can do something.

shampeon, 2020-04-12:
I looked at his fork, and yeah, it absolutely breaks HA integration. 

It ignores the Fahrenheit/Celsius handling code I added, and only in his internal changes, so e.g. you can set the unit to Fahrenheit in the web interface, but his code ignores it when setting temperature. That means that if you manually set the unit to F and set the temperature to 68 degrees in the web interface, 68 degrees Celsius will be sent to the Swicago library, which obviously won't work. So his fork has broken temperature support.

It removes the status, so you can't tell if the unit is idle or doing something else.

I also really have to repeat that the Swicago example code for MQTT isn't a standard and isn't a feature of his library, and I don't think Swicago even uses MQTT with his units.

I'm fine if he or someone else wants to add an option to enable HA/Domoticz/whatever support, with conditionals for the MQTT topic handling. But HA autodiscovery requires separate set and state topics for the mode, fan, temperature, power, step, and min/max temperature, so HA can't just use the single set and state topics like the Swicago MQTT example code.

gysmo38, 2020-04-13:
@shampeon I am agree with you.
@spfmoby I try the plugin. 
I found two problems:
- I can not set login and password for MQTT
- The plugin implement the MQTT part and not user the implementation made by domoticz
I think a best way is to use domoticz mqtt integration to do it.

uronito, 2020-04-15:
@gysmo38, @spfmoby  with your permission I have forked the domoticz implementation and removed the has implementation. It works by following the guidelines of the swqago mqtt example, in addition http control is enabled without the need to configure mqttt. It works perfect with the domoticz plugin made in python for mitsubishi. Throughout the day I will upload the files to my fork, domoticz branch.
Regards.

ianjevans, 2020-04-15:
So I took a look at the Masurov Domoticz-MitsubishiHpMQTT-Plugin code again, and as far as I can tell, it would work with our mitsubishi2MQTT code with just 2 or 3 tiny code changes.

https://github.com/Masurov/Domoticz-MitsubishiHpMQTT-Plugin/blob/master/plugin.py#L13
Set Mode5 parameter to "mitsubishi2mqtt". 
`<param field="Mode5" label="Heatpump MQTT topic" width="300px" required="true" default="mitsubishi2mqtt"/>`

I imagine this parameter can be set within Domoticz, though I don't use it so I don't know for certain.

https://github.com/Masurov/Domoticz-MitsubishiHpMQTT-Plugin/blob/master/plugin.py#L171
Set self.HeatpumpSetTopic to "settings":
`self.HeatpumpSetTopic = self.HeatpumpTopic + "/settings"`

https://github.com/Masurov/Domoticz-MitsubishiHpMQTT-Plugin/blob/master/plugin.py#L172
Set RoomTempTopic to "state":
`self.RoomTempTopic = self.HeatpumpTopic + "/state"`

spfmoby, 2020-04-16:
ianjevans
 The first parameter is freely customizable in Domoticz, so it is not a real problem.
For the 2 others, I tried to edit the plugin on my domoticz instance and it does not seem enough to make it work.
```
            #self.HeatpumpSetTopic = self.HeatpumpTopic + "/set"
            self.HeatpumpSetTopic = "settings"

            #self.RoomTempTopic = self.HeatpumpTopic + "/status"
            self.RoomTempTopic = "state"

```
Is it what you mean by  Set self.HeatpumpSetTopic to "settings" ? or should we just replace "set" by "settings" and  "status" by "state" on the whole line ?

shampeon, 2020-04-16:
I meant that lines 171 and 172 should be:
```
self.HeatpumpSetTopic = self.HeatpumpTopic + "/settings"
self.RoomTempTopic = self.HeatpumpTopic + "/state"
```

This is to match the settings and state topics in mitsubishi2MQTT. E.g.
```
ian@diablo:~ $ mosquitto_sub -h x.x.x.x -v -t "mitsubishi2mqtt/playroom/#"
mitsubishi2mqtt/playroom/settings {"temperature":70,"fan":"AUTO","vane":"AUTO","mode":"off"}
mitsubishi2mqtt/playroom/state {"roomTemperature":67,"temperature":70,"fan":"AUTO","vane":"AUTO","action":"off","mode":"off"}
```

Swicago's MQTT example uses "set" and "status".

shampeon, 2020-04-16:
And I just noticed another limitation of Swicago's MQTT example: the MQTT topic only works with a single heatpump.

So for each Domoticz heatpump, you'll need to set Mode5 to the unit name you set in mitsubishi2MQTT. E.g. if the "Friendly Name" you set on setup is "foo", set Mode5 to "mitsubishi2mqtt/foo"

spfmoby, 2020-04-16:
OK, thanks for the details.
In Mitsubishi2MQTT I've set:
topic : "mitsubishi2mqtt"
friendly name : "playroom" (to stick to your example)

In the Domoticz plugin I modified the lines 171 and 172 to:
self.HeatpumpSetTopic = self.HeatpumpTopic + "/settings"
self.RoomTempTopic = self.HeatpumpTopic + "/state"

In the parameters of the plugin:
mqtt topic: mitsubishi2mqtt/playroom

And... it works... halfway for now. I mean that the "state" functions seems to work properly. I now have the temperature in domoticz, and if I change a parameter in Mitsubishi2MQTT like the fan speed or on the AC remote, it is passed on to domoticz. But:
- the power status, modes... does not seems to be reported properly in domoticz
- the "set" functions in domoticz don't work. (I can't turn on or change the fan speed)

shampeon, 2020-04-18:
Ok, thanks for checking. I think the modes aren't quite the same between mitsubishi2mqtt and the Domoticz plugin, so even though the mode is being reported in the state/settings topic it's not being picked up by the Domoticz plugin.

There's 3 ways we can fix this, I think, in ascending order of complexity:
- Harmonize the mode/set names in mitsubishi2mqtt.
- Add support for the reading/writing to the mode/fan/vane topics in the Domoticz plugin.
- Add a Domoticz support mode to mitsubishi2mqtt to read/write to the MQTT topics expected by the current Domoticz plugin.

gysmo38, 2020-06-28:
No news. I close this issue. Re open or open new one if needed.

closed 2020-06-28 by gysmo38