gysmo38/mitsubishi2MQTT #45: Upload to board fails

Turbobussi, 2020-04-09:
Sketch uses 290516 bytes (27%) of program storage space. Maximum is 1044464 bytes.
Global variables use 29440 bytes (35%) of dynamic memory, leaving 52480 bytes for local variables. Maximum is 81920 bytes.
esptool.py v2.8
Serial port COM5
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: ec:fa:bc:5e:ca:1a
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 294672 bytes to 213918...

Writing at 0x00000000... (7 %)
Writing at 0x00004000... (14 %)
Writing at 0x00008000... (21 %)
Writing at 0x0000c000... (28 %)
Writing at 0x00010000... (35 %)
Writing at 0x00014000... (42 %)
Writing at 0x00018000... (50 %)
Writing at 0x0001c000... (57 %)
Writing at 0x00020000... (64 %)
Writing at 0x00024000... (71 %)
Writing at 0x00028000... (78 %)
Writing at 0x0002c000... (85 %)
Writing at 0x00030000... (92 %)
Writing at 0x00034000... (100 %)
Wrote 294672 bytes (213918 compressed) at 0x00000000 in 4.8 seconds (effective 489.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Invalid library found in arduino\libraries\mitsubishi2MQTT-master: no headers files (.h) found in arduino\libraries\mitsubishi2MQTT-master
Invalid library found in arduino\libraries\mitsubishi2MQTT-master: no headers files (.h) found in arduino\libraries\mitsubishi2MQTT-master

After that no wifi appears.

brusarp, 2020-04-14:
I also use a D1 mini card and it works smoothly ... it's strange

gysmo38, 2020-04-14:
@Turbobussi Before upload, in flash size check you have FS size at 64KB or more

Turbobussi, 2020-04-14:
Are these correct?

[image]

gysmo38, 2020-04-14:
For first upload try to but Erase Flash to All Flash contents. After for upgrade it look good

Turbobussi, 2020-04-14:
Thanks. That worked!

closed 2020-04-14 by Turbobussi

gysmo38, 2020-04-14:
:) do not forget to change it for upgrading or you will loose all your configuration