gysmo38/mitsubishi2MQTT #52: Sorry for asking about this - programming help:)

jascdk, 2020-04-15:
First of all - I know this is way off for this project BUT - I think this project has turned out to be very nice - and with a very cool interface / GUI. 

I have a project going on with controlling a Danish branded Nilan Vent system, that is controlled via modbus. the project is here https://github.com/jascdk/Nilan_Homeassistant

I really want to have some kind of same GUI experience like in this project, but the code is massive and I simply don't know where to start (im not much into coding :/) 

Does someone of you have a kind of template / thing for this code, so I can get on going. I have tried to pick out parts and so, but it always messes up :( 

Thanks in advance and sorry for posting this "issue" as it is not related to this project.

All the best

Jacob

gysmo38, 2020-04-23:
Hello,
Sorry for late answer...
I use <ESP8266WebServer.h>  for webserver
You can find tutorials to understand how it work.
For mitshubishi2mqtt code, all HTML code are in hearder file. For example html_pages.h have body html code, html_common have header and footer html code.
In html code we use _VARIABLES_ string to have dynamic content. We replace it in C code.
Have a look at the function handleStatus() in mitsubishi2mqtt.ino to have an example.

jascdk, 2020-04-23:
@gysmo38 thats alright :D Big thanks - will try checking it out:)

gysmo38, 2020-04-28:
@jascdk I close the issue but feel free to re open or open a new one if needed.

closed 2020-04-28 by gysmo38