gysmo38/mitsubishi2MQTT #53: last action stays in the url

spfmoby, 2020-04-17:
Hi
Something is bothering me : I often keep a tab open with the control page of my AC (ie : http://192.168.1.33/control ) When I make an action, like turning on or off the AC, the url becomes http://192.168.1.33/control?POWER=OFF for example.
If later I refresh (or the tab is automatically refreshed after I quit and restarted my browser) and in the meantime I changed the status of the AC (it is powered on for example), it automatically shuts it off which is not very convenient.
Is there a reason for the url to be keeping the last action? Wouldn't it be better if a new location was called (/control) after each command or the url rewritten using javascript for example?

shampeon, 2020-04-18:
Changing the method to POST should keep the params from the URL.

closed 2020-04-18 by gysmo38

spfmoby, 2020-04-19:
thanks. I tried, it works, except for the temperature which is still a POST request

commit 11182a91 (2020-04-18): Changing control forms to use POST to fix #53.