gysmo38/mitsubishi2MQTT #58: Loading control page without MQTT configured causes board to reset

wabarkley, 2020-04-22:
As also noted by @andrewl3wis in the gitter, when I first flashed my boards I attempted to load the control page and the board would reset - sending a wdt error through the serial connection.

Once I filled out the MQTT settings, the control page started working correctly. I have six boards total and all are working perfectly with MQTT enabled, I verified the board reset issue on the last one I installed before configuring MQTT. Once configured, the control page loads fine.

gysmo38, 2020-04-28:
ok thank. I will check if it possible to add a check before load the page

wabarkley, 2020-04-29:
Thanks. Ideally, the control page would load and operate correctly even if mqtt wasn't configured. 
This repo really is super easy to use and fairly robust already. For folks that aren't necessarily interested in mqtt, this would be great to just control the heat pump.

rscoleNZ, 2020-10-10:
I wish I had seen this... As a complete beginner I assumed either my soldering was wrong tx/rx won't way round, or I was about to reflash my board until I checked here.
But I agree, having it work without MQTT would be ideal.
Thanks for everything you've done to date.

smichtch, 2020-11-23:
I ran into this as well and was convinced that I botched the crimp terminals yet again.

I think what's happening is `handleControl()` calls `change_states(settings)` which triggers `hpSettingsChanged()` and/or `hpStatusChanged()` callbacks and neither of those test whether `mqtt_client` is initialized before trying to publish and that must be causing the crash/reset?
Also `handleControl()` should probably skip calling `change_states(settings)` on GET requests.

This is just speculation after skimming though the code, haven't had a chance to try to fix the code yet.

wabarkley, 2020-11-23:
Do you think it would be more efficient to initialize the mqtt client but keep it disconnected in some way, or to wrap all the various publish commands with a mqtt_client initialization check?

wabarkley, 2020-11-23:
Digging in further, I'm thinking about just setting the `setSettingsChangedCallback` and `setStatusChangedCallback` callbacks within `initMQTT` since I know both `hp` and `mqtt_client` will have been initialized at that point. (I'm not sure if that's bad practice, though.)

smichtch, 2020-11-24:
Ok my hypothesis about `hpSettingsChanged()`/`hpStatusChanged()` was wrong.
It's actually crashing in `handleControl()` when accessing the `heatpumpSettings` struct:
```
  if (strcmp(settings.power, "ON") == 0) {
    controlPage.replace("_POWER_ON_", "selected");
  }
  ...
```
Looks like the default `heatpumpSettings` is corrupt when `loop()` isn't polling the heat pump - i.e. when MQTT is not configured.
The control page seems happy after changing `loop()` from:
```
  if (!captive and mqtt_config) {
    // Sync HVAC UNIT
    ...
    //MQTT failed retry to connect
    ...
  }
  else ...
```
to
```
  if (!captive) {
    // Sync HVAC UNIT
    ...
    if (mqtt_config) {
      //MQTT failed retry to connect
      ...
    }
  }
  else ...
```

PR #119 (2020-12-02): Control page without mqtt
Implemented @smichtch suggestion inside loop to allow control page to bypass MQTT code when MQTT isn't configured. This Resolves #58 .
closed 2020-12-02 by gysmo38