gysmo38/mitsubishi2MQTT #63: Use platformio for development

jmvaz, 2020-04-29:
In order to avoid the problems with the installation of all dependencies, we should prepare the project for platformio.
Once you try it you will never return to ArduinoIDE :)

lesleypersyn, 2020-04-29:
In the past i did created the platformio.ini but it's not usable anymore due to many changes i did. (and the current project is further developed)

We should normally create this side by side the arduinoIDE.

i have just some point where i'm stuck:
- ino vs cpp extension of mitsubishimqtt file (cannot use ino files in platformio, should be converted/renamed to cpp. Correct me if i'm wrong)
- many errors about the functions created in mitsubishi2mqtt
- ...

gysmo38, 2020-06-28:
I think it's done :)

closed 2020-06-28 by gysmo38