gysmo38/mitsubishi2MQTT #67: Never reverts to normal operation after startup with no Wifi signal

madzur, 2020-05-12:
Hello Gysmo38,

first of all, big props on this development. I have been following SwiCago and you for over a year and I love what you have done with the autodiscovery code and how reliably it now works with HASS.

There is however one quirk I have been having issues with, since I live in an area with frequent power outages. Whenever a power outage happens, all my electronic equipment boots up at the same time. Heatpumps are quite quick and boot up before my Wifi APs. Therefore they can't find the WiFi signal and revert to AP mode with the HVAC_XXXX ssid.

However, when I try to fix this, I power-cycle the heatpumps manually, with WiFi signal already available, the heatpumps stay in the HVAC_XXXX mode indefinitely.
To fix this, I have to connect to each of my 3 HVAC_XXXX APs, I have to remember which MAC is used in which heatpump, I have to provide the WiFi SSID and Password and after restarting each module, the normal operation resumes. The module remembers MQTT details, temperature step settings and all other customizations, so the issue seems to be only with the initial WiFi config.

What I would like to happen instead, is that when a manual power cycle is triggered, upon rebooting and finding the known WiFi network, normal operation would be resumed.

Do you know what I mean?

gysmo38, 2020-05-12:
I think this is a bug.
I think when we have wifi outage we go to intial config but we do not test if the WIFI come back.
Not sure I will made some tests.
If you wand to save time, you can just click on reboot button on the inital conf. It will reboot your esp and connect again to the wifi.

madzur, 2020-05-12:
Thanks for your reply. I will test this and report back.
One thing I noticed, when entering the WiFi details again, my Heatpump controls in HASS were left unavailable. I had too look into the entities list and found my heatpumps with "_2" in the list.

Therefore I removed my original lovelace cards used to control heatpumps and added the new "_2" entities. Also seems strange behavior. Just wanted to let you know.

juampe, 2020-05-31:
I have the same issue when the main went off. I can see my 3 HVAC_XXXX devices in the same state. I can connect to the captive portal, and now it haves a password, same as AP name. I can set ESSID and password, all now all works again. The "reconfiguration" keeps the mqtt configuration. Maybe we must check if there are a configuration stored in order to WiFi retry?

juampe, 2020-06-01:
This patch might help waiting 300 seconds AP 

```
diff --git a/src/mitsubishi2mqtt/mitsubishi2mqtt.ino b/src/mitsubishi2mqtt/mitsubishi2mqtt.ino
index d5096a6..7e706e9 100644
--- a/src/mitsubishi2mqtt/mitsubishi2mqtt.ino
+++ b/src/mitsubishi2mqtt/mitsubishi2mqtt.ino
@@ -1604,7 +1604,7 @@ bool connectWifi() {
   WiFi.begin(ap_ssid.c_str(), ap_pwd.c_str());
   // Serial.println("Connecting to " + ap_ssid);
   unsigned long wifiStartTime = millis();
-  while (WiFi.status() != WL_CONNECTED && millis() - wifiStartTime < 10000) {
+  while (WiFi.status() != WL_CONNECTED && millis() - wifiStartTime < 300000) {
     Serial.write('.');
     //Serial.print(WiFi.status());
     // wait 500ms, flashing the blue LED to indicate WiFi connecting...
```

This patch might help, trying forever the configuration previusly stored in SPIFS (same behaviour as tasmota)

```
diff --git a/src/mitsubishi2mqtt/mitsubishi2mqtt.ino b/src/mitsubishi2mqtt/mitsubishi2mqtt.ino
index d5096a6..5619a66 100644
--- a/src/mitsubishi2mqtt/mitsubishi2mqtt.ino
+++ b/src/mitsubishi2mqtt/mitsubishi2mqtt.ino
@@ -1603,8 +1603,7 @@ bool connectWifi() {
 #endif
   WiFi.begin(ap_ssid.c_str(), ap_pwd.c_str());
   // Serial.println("Connecting to " + ap_ssid);
-  unsigned long wifiStartTime = millis();
-  while (WiFi.status() != WL_CONNECTED && millis() - wifiStartTime < 10000) {
+  while (WiFi.status() != WL_CONNECTED) {
     Serial.write('.');
     //Serial.print(WiFi.status());
     // wait 500ms, flashing the blue LED to indicate WiFi connecting...
@@ -1622,7 +1621,7 @@ bool connectWifi() {
   // Serial.println(F("Ready"));
   // Serial.print("IP address: ");
   unsigned long dhcpStartTime = millis();
-  while ((WiFi.localIP().toString() == "0.0.0.0" || WiFi.localIP().toString() == "") && millis() - dhcpStartTime < 5000) {
+  while (WiFi.localIP().toString() == "0.0.0.0" || WiFi.localIP().toString() == "") {
     // Serial.write('.');
     delay(500);
   }
```

2920, 2020-06-04:
> I think this is a bug.
> I think when we have wifi outage we go to intial config but we do not test if the WIFI come back.
> Not sure I will made some tests.
> If you wand to save time, you can just click on reboot button on the inital conf. It will reboot your esp and connect again to the wifi.

It's not a bug, there was never a feature to reconnect. Enhancement required.

gysmo38, 2020-06-05:
@2920  You're right this is not a bug. An improvement could be to have a menu with 2 choice : Reboot / Reconfigure when there is already configuration file.

loadrunner42, 2020-09-20:
This implementation will auto reboot when the wifi has dropped or in AP mode after 5min and there is already a configuration file:

diff --git a/src/mitsubishi2mqtt/config.h b/src/mitsubishi2mqtt/config.h
index 23e82bb..435d6df 100644
--- a/src/mitsubishi2mqtt/config.h
+++ b/src/mitsubishi2mqtt/config.h
@@ -40,6 +40,9 @@ const PROGMEM  uint8_t redLedPin = 0;
 
 // Define global variables for network
 const PROGMEM char* hostnamePrefix = "HVAC_";
+const PROGMEM uint32_t WIFI_RETRY_INTERVAL_MS = 300000;
+int wifi_timeout;
+bool wifi_config_exists;
 String hostname = "";
 String ap_ssid;
 String ap_pwd;
diff --git a/src/mitsubishi2mqtt/mitsubishi2mqtt.ino b/src/mitsubishi2mqtt/mitsubishi2mqtt.ino
index 49e2531..5ed859a 100644
--- a/src/mitsubishi2mqtt/mitsubishi2mqtt.ino
+++ b/src/mitsubishi2mqtt/mitsubishi2mqtt.ino
@@ -110,7 +110,7 @@ void setup() {
   WiFi.hostname(hostname.c_str());
 #endif
   setDefaults();
-  loadWifi();
+  wifi_config_exists = loadWifi();
   loadOthers();
   loadUnit();
   if (initWifi()) {
@@ -500,10 +500,11 @@ boolean initWifi() {
 
   // Serial.println(F("\n\r \n\rStarting in AP mode"));
   WiFi.mode(WIFI_AP);
+  wifi_timeout = millis() + WIFI_RETRY_INTERVAL_MS;
   WiFi.persistent(false); //fix crash esp32 https://github.com/espressif/arduino-esp32/issues/2025
-  if (!connectWifiSuccess) {
+  if (!connectWifiSuccess and login_password != "") {
     // Set AP password when falling back to AP on fail
-    WiFi.softAP(hostname.c_str(), hostname.c_str());
+    WiFi.softAP(hostname.c_str(), login_password);
   }
   else {
     // First time setup does not require password
@@ -1624,7 +1625,8 @@ bool connectWifi() {
 #endif
   WiFi.begin(ap_ssid.c_str(), ap_pwd.c_str());
   // Serial.println("Connecting to " + ap_ssid);
-  while (WiFi.status() != WL_CONNECTED) {
+  wifi_timeout = millis() + 30000;
+  while (WiFi.status() != WL_CONNECTED && millis() < wifi_timeout) {
     Serial.write('.');
     //Serial.print(WiFi.status());
     // wait 500ms, flashing the blue LED to indicate WiFi connecting...
@@ -1734,6 +1736,14 @@ void checkLogin() {
 void loop() {
   server.handleClient();
   ArduinoOTA.handle();
+  
+  //reset board to attempt to connect to wifi again if in ap mode or wifi dropped out and time limit passed
+  if (WiFi.getMode() == WIFI_STA and WiFi.status() == WL_CONNECTED) {
+	  wifi_timeout = millis() + WIFI_RETRY_INTERVAL_MS;
+  } else if (wifi_config_exists and millis() > wifi_timeout) {
+	  ESP.restart();
+  }
+  
   if (!captive and mqtt_config) {
     // Sync HVAC UNIT
     if (!hp.isConnected()) {
--

gysmo38, 2021-02-24:
I think it is resolve with PR #104

closed 2021-02-24 by gysmo38