gysmo38/mitsubishi2MQTT #75: Help to explain some MQTT commands

picotouch, 2020-05-25:
HI,
in Readme the last chapter contains a list of topics.
Please explain some of them (bold expressions).

- topic/power/set OFF	                                                                ==> switch OFF unit, working OK
- topic/mode/set AUTO HEAT COOL DRY FAN_ONLY OFF ON	==> set working mode, working OK. Thanks for ON command!!!
- topic/temp/set 16-31	==> set temperature, working OK
- **topic/remote_temp/set**	==> what is this and how to use it? Example...?
- topic/fan/set 1-4 AUTO QUIET	==> set fan mode, working OK
- topic/vane/set 1-5 SWING AUTO	==> set vane mode, working OK
- topic/wideVane/set << < | > >>	==> set wide vane mode, my model does not have this, I guess it's OK.
- topic/settings	==>  response from air conditioner, working OK
- topic/state	==>  state from  air conditioner, working OK
- topic/debug	==>  show debuged data, working OK
- **topic/debug/set** on off	==>  Not working if it is set from MQTT, working if it is set from web portal
- **topic/custom/send**	==> what is this and how to use it? Example...?

And finally, if on the web portal in Setup / Others, I set HA Autodisovery to OFF, after save & reboot, the state is ON again?
I don't use HA, but is this a bug?

Thanks!!

juampe, 2020-05-31:
AFAIK 
topic/remote_temp/set is for external thermometer, it maps the room_temp parameter from the  original library, in theory you can set it from another sensor. It seems not work in my MSZ-HR.

topic/debug/set I can see the _debugMode in the code but, but it does not publish the "Debug mode enabled/disables/message" as i set to on or off to the topic, "it seems a bug".

topic/custom/send is for send raw packets, see https://github.com/SwiCago/HeatPump/blob/master/src/HeatPump.h , it needs knoledge of the raw protocol.

juampe, 2020-06-02:
About the topic/debug/set I have to made little change to see why debug does not work. I am not familiarized with the F function (RTFM for me), but if I remove it, and the code works as expected, and mqtt debug is operational. I compile with PIO ESP8366 2.5.1

```
diff --git a/src/mitsubishi2mqtt/mitsubishi2mqtt.ino b/src/mitsubishi2mqtt/mitsubishi2mqtt.ino
index 5619a66..a308706 100644
--- a/src/mitsubishi2mqtt/mitsubishi2mqtt.ino
+++ b/src/mitsubishi2mqtt/mitsubishi2mqtt.ino
@@ -1433,10 +1433,10 @@ void mqttCallback(char* topic, byte* payload, unsigned int length) {
   else if (strcmp(topic, ha_debug_set_topic.c_str()) == 0) { //if the incoming message is on the heatpump_debug_set_topic topic...
     if (strcmp(message, "on") == 0) {
       _debugMode = true;
-      mqtt_client.publish(ha_debug_topic.c_str(), (char*)(F("Debug mode enabled")));
+      mqtt_client.publish(ha_debug_topic.c_str(), (char*)"Debug mode enabled");
     } else if (strcmp(message, "off") == 0) {
       _debugMode = false;
-      mqtt_client.publish(ha_debug_topic.c_str(), (char*)(F("Debug mode disabled")));
+      mqtt_client.publish(ha_debug_topic.c_str(), (char*)"Debug mode disabled");
     }
   }
   else if(strcmp(topic, ha_custom_packet.c_str()) == 0) { //send custom packet for advance user
```

juampe, 2020-06-04:
Seems that we found a 32 bit alignment problem, related to the use of the F() macro in this code section
https://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html

> The two examples above will store these strings in flash. To retrieve and manipulate flash strings they must be read from flash in 4byte words. In the Arduino IDE for esp8266 there are several functions that can help retrieve strings from flash that have been stored using PROGMEM. Both of the examples above return const char *. However use of these pointers, without correct 32bit alignment you will cause a segmentation fault and the ESP8266 will crash. You must read from the flash 32 bit aligned.

The publish function seems to handle F() correctly 
https://imroy.github.io/pubsubclient/classMQTT_1_1Publish.html

> Publish (String topic, const __FlashStringHelper *payload)
> Constructor from a string stored in flash using the F() macro.

picotouch, 2020-06-13:
@juampe, thanks for explaining and updating Readme.md!

closed 2020-06-13 by picotouch