gysmo38/mitsubishi2MQTT #79: Get current temperature?

andig, 2020-06-02:
If the MEL is able to work against a target temperature, it should be able to understand the current room temperature. I'm wondering if that value is available anywhere?

juampe, 2020-06-02:
Hi,
You can collect the status by subscribe to a MQTT topic json in this topic, and is called "roomTemperature"
The topic is:
`topic/HVAC_XXX/state `
The json received:
`{"roomTemperature":26.5,"temperature":26,"fan":"AUTO","vane":"AUTO","action":"cooling","mode":"cool"}`

andig, 2020-06-02:
Great- might be nice to add those to the readme or even expose those JSON values as subtopics to `state` for easier consumption?

gysmo38, 2020-06-05:
In fact it is already the case you have two topics: settings and state

andig, 2020-06-05:
> In fact it is already the case you have two topics: settings and state

I don't think so- or its not documented (sorry, still haven't installed my adapter). I'm looking for the *current* temperature, something like

    topic/HVAC_XXX/state/temperature

That doesn't seem to exist yet?

adamderuwe, 2020-06-15:
[image]

andig, 2020-06-16:
@adamderuwe I may not have made this clear enough. What I meant is

    HVAC_master/state/roomTemperature
    HVAC_master/state/temperature
    HVAC_master/state/fan

etc. That way one could use the state without further need for decoding the json!

wabarkley, 2020-11-24:
It seems fairly common to arrange the state the way the library implements and MQTT Explorer is able to parse out the JSON. HomeAssistant also is working fine with the existing code.

I have seen another MQTT "standard" proposed that arranges them the way you're looking for at https://homieiot.github.io/tools/.

Arranged something along the lines of...

homie/mitsubishi2MQTT/$homie 4.0
homie/mitsubishi2MQTT/$name My device
homie/mitsubishi2MQTT/$state ready
homie/mitsubishi2MQTT/$extensions 
homie/mitsubishi2MQTT/$nodes mythermostat

homie/mitsubishi2MQTT/mythermostat/$name Thermostat 1
homie/mitsubishi2MQTT/mythermostat/$properties roomTemperature
homie/mitsubishi2MQTT/mythermostat/$properties temperature
homie/mitsubishi2MQTT/mythermostat/$properties fan
homie/mitsubishi2MQTT/mythermostat/$properties vane
homie/mitsubishi2MQTT/mythermostat/$properties wideVane
homie/mitsubishi2MQTT/mythermostat/$properties mode
homie/mitsubishi2MQTT/mythermostat/$properties action
homie/mitsubishi2MQTT/mythermostat/$properties compressorFrequency

homie/mitsubishi2MQTT/mythermostat/temperature 22 
homie/mitsubishi2MQTT/mythermostat/temperature/$name Temperature
homie/mitsubishi2MQTT/mythermostat/temperature/$unit °C
homie/mitsubishi2MQTT/mythermostat/temperature/$datatype integer
homie/mitsubishi2MQTT/mythermostat/temperature/$settable integer

closed 2022-01-16 by gysmo38