gysmo38/mitsubishi2MQTT #83: add an I/O Pin for a door/window contact

Eisbahn, 2020-06-22:
What about using one (or two) of the unused Pins as Input pin for a door or window contact (as the AC is quite often build on top of a window or door) and transmit the status change of this contact on change? That would allow e.g. stopping the AC if the door/window is open for more than 5min or similar use cases...

Eisbahn, 2020-07-17:
switched to ESPeasy, the testing one has a full working implementation in it and gives several benefits:
- update only on changed values and no fixed cyclic spamming from aircon
- easy to have a door contact, status LED, presence/radar sensor, humidity device... added
- OTA feature
- big dev group with frequnet updates
No need for me to further develop here anything, could be closed.

gysmo38, 2020-07-17:
Happy you find a solution for your need :)

closed 2020-07-17 by gysmo38