gysmo38/mitsubishi2MQTT #84: deactivation of HomeAssistant does not work from Web-Menue

Eisbahn, 2020-06-24:
I'll try to deactivate the HomeAssistant feature (as it is useless for me) via WebMenue by setting it to "OFF" and afterwards clicking on Save & Reboot. After 10sec and the reconnect, HA is still "ON" in the web and also if I sniff the messages it is still active

Eisbahn, 2020-06-24:
just had a quick look into the code:
as others_haa is by default set to true, the "off" case should be handled from my opinion around line 468...470
`  if (strcmp(haa.c_str(), "ON") == 0) {
    others_haa = true;
  }
`
to somthing like
`  
  if (strcmp(haa.c_str(), "ON") == 0) {
    others_haa = true;
  }
else {
others_haa = false;
`

gysmo38, 2020-06-28:
Done thank you :)

closed 2020-06-28 by gysmo38