gysmo38/mitsubishi2MQTT #88: MQTT availability feature

giuseppeg88, 2020-06-30:
Can implement the MQTT availability feature?

[From Home Assistant docs](https://www.home-assistant.io/integrations/binary_sensor.mqtt/):

**_availability_topic_**
_The MQTT topic subscribed to receive birth and LWT messages from the MQTT device. If availability_topic is not defined, the binary sensor will always be considered available and its state will be on, off or unknown. If availability_topic is defined, the binary sensor will be considered as unavailable by default and the sensor’s state will be unavailable._

matejzero, 2020-09-20:
This would be a really nice feature, as right now, there is no feedback weather the AC is accessible or not.

I have an AC with a bit of a poor signal sometimes and with this, I could see right away if it dropped from network.

gysmo38, 2022-01-16:
I valide a PR for that. It should work. If not open a new issue.

closed 2022-01-16 by gysmo38