gysmo38/mitsubishi2MQTT #96: Wide vane setting doesn't show up in topic/settings or topic/state

melyux, 2020-08-17:
Sorry about the rapid-fire issues, but I'm filing as I try the firmware out. My Mitsubishi supports the wide vane setting, and setting it works really well through the Mitsbuishi2MQTT web server. But the wideVane parameter just isn't present in the `topic/settings` or `topic/state` topics, so inaccessible to the MQTT broker (and so, for example, HA).

I'll dive into the code and see if I can spot why this is not working in the meantime.

stenleo, 2020-08-19:
Seconding this, good to hear someone is looking into it!

mamrai1, 2020-08-31:
but mqtt climate component of home assistant does not support widevane...

stenleo, 2020-08-31:
> but mqtt climate component of home assistant does not support widevane...

That might be the case. Couldn't it be possible to just expose the functionality via standard mqtt. That is a good enough solution for me, just setting a value to mqtt. The MelCloud integration has support, they add extra services for the wide vane. This is the only functionality keeping me from tossing out the MelCloud integration right now.

melyux, 2020-08-31:
> but mqtt climate component of home assistant does not support widevane...

You can just make a separate “fan” component in HA to use this new capability for now.

@stenleo Have you tried my PR?

mamrai1, 2020-09-01:
> > but mqtt climate component of home assistant does not support widevane...
> 
> You can just make a separate “fan” component in HA to use this new capability for now.
> 
> @stenleo Have you tried my PR?

I tried but it doesn't work.. I think you didn't add json values for widevane

mamrai1, 2020-09-01:
lines  after 1528 ..you need to add for widevane i think:
  JsonArray haConfigSwing_modes = haConfig.createNestedArray("swing_modes");
  haConfigSwing_modes.add("AUTO");
  haConfigSwing_modes.add("1");
  haConfigSwing_modes.add("2");
  haConfigSwing_modes.add("3");
  haConfigSwing_modes.add("4");
  haConfigSwing_modes.add("5");
  haConfigSwing_modes.add("SWING");

  haConfig["swing_mode_cmd_t"]              = ha_vane_set_topic;
  haConfig["swing_mode_stat_t"]             = ha_state_topic;
  haConfig["swing_mode_stat_tpl"]           = F("{{ value_json.vane if (value_json is defined and value_json.vane is defined and value_json.vane|length) else 'AUTO' }}"); //Set default value for fix "Could not parse data for HA"
  haConfig["action_topic"]                  = ha_state_topic;
  haConfig["action_template"]               = F("{{ value_json.action if (value_json is defined and value_json.action is defined and value_json.action|length) else 'idle' }}"); //Set default value for fix "Could not parse data for HA"

stenleo, 2020-09-01:
> > but mqtt climate component of home assistant does not support widevane...
> 
> You can just make a separate “fan” component in HA to use this new capability for now.
> 
> @stenleo Have you tried my PR?

Haven't tried it, is it stable otherwise? I can give it a try, but considering mamrai1's comment I'm a bit hesitant. Does it still need some additional fixes?
Otherwise I am all for creating a separate component for this functionality in HA, that is definitely good enough.

closed 2020-09-03 by gysmo38

mamrai1, 2020-09-03:
> Closed #96 <https://github.com/gysmo38/mitsubishi2MQTT/issues/96> via #97

Please check again... It's not ok

melyux, 2020-09-03:
@mamrai1 We don't want to add the wide vane settings to the HA config, since the HA climate component doesn't support vertical swing. I didn't add to the `haConfig` on purpose. You have to manually create an MQTT entity in your HA config to integrate the wide vane.

mamrai1, 2020-09-03:
@melyux yes you are right but something is wrong. Compilation fails

melyux, 2020-09-03:
@mamrai1 Thanks for fixing it with #100! You were right.

melyux, 2020-09-03:
I've tested @mamrai1's #101 PR on my unit and it's working very well so far.

stenleo, 2020-09-09:
Perfect, thanks for the work on this. See it's been merged with the main branch, I'll give that a try!

Knot83, 2023-07-19:
Guys I’m in the situation where I can see both vane options in the web page but only one option in Home Assistant. I read in this post that Mqtt in HA doesn’t support wide vane but I didn’t get how to workaround this problem. Can someone explain me step by step what to do?

mamrai1, 2023-07-19:
My code adjusted for tasmota miel driver:

input_select:
  living_ac_widevane:
    name: WideVane
    options:
       -  "swing"
       -  "LL"
       -  "L"
       -  "C"
       -  "R"
       -  "RR"   
       -  "split"       
    icon: mdi:target

automation:
- alias: "Set Living AC Widevane Selector"
  trigger:
    platform: mqtt
    topic: "tele/tasmota_AD9FA1/HVACSETTINGS"
  action:
    service: input_select.select_option
    target:
      entity_id: input_select.living_ac_widevane
    data:
      option: "{{ trigger.payload_json.SwingH }}"
  mode: restart
- alias: "Set Living AC Widevane"
  trigger:
    platform: state
    entity_id: input_select.living_ac_widevane
  action:
    service: mqtt.publish
    data:
      topic: "cmnd/tasmota_AD9FA1/HVACSetSwingH"
      retain: false
      payload: "{{ states('input_select.living_ac_widevane') }}"

Knot83, 2023-07-19:
> My code adjusted for tasmota miel driver:
>
> input_select: living_ac_widevane: name: WideVane options: - "swing" - "LL" - "L" - "C" - "R" - "RR" - "split" icon: mdi:target
>
> automation:
>
> * alias: "Set [...]

So If I understood correctly I'll have to create an Input Select and an Automation that Publish the MQTT topic based on what I select in the Input Select for the vane. Then I'll ad the input select in the GUI. Right? Can you also help me to understand where to sniff the name of the MQTT topi for the wide vane?

mamrai1, 2023-07-19:
ADJUSTED:

input_select:
living_ac_widevane:
name: WideVane
options:
- "<< "
- "<"
- "|"
- ">"
- ">>"
- "<>"
- "SWING"
icon: mdi:target

Knot83, 2023-07-20:
@mamrai1 Thanks for you support I've been able to implement also the Wide Vane. Now I've a doubt. Since I didn't have another D1 Mini for my second AC, I flashed this firmware in a NodeMCU but I'm having some doubts if this firmware is ok with the pin configuration of a NodeMCU that is different from a D1 Mini. Any idea? So far I can see correctly the Web Interface but I'm a bit scared to connect it to the AC. Thanks

mamrai1, 2023-07-20:
> @mamrai1 Thanks for you support I've been able to implement also the Wide Vane. Now I've a doubt. Since I didn't have another D1 Mini for my second AC, I flashed this firmware in a NodeMCU but I'm [...]

The firmware use GPIO3/GPIO1 so there is no difference

Knot83, 2023-07-20:
Yea correct it works as well! Thanks again! Have a great day!

Knot83, 2023-07-20:
Last question: is there any specific reason why Wide vane doesn’t appear in the native climate entity of HA? Is there a way to add it directly in the climate entity?

mamrai1, 2023-07-20:
No it's not supported. Personally i use simple thermostat card from hacs. 
[image]
[image]

mamrai1, 2025-02-26:
MQTT climate component will support in next release swing horizontal mode:
https://github.com/home-assistant/core/pull/139303

PR #97 (2020-09-03): Add missing wide vane MQTT logic
Wide vane was implemented on the web server, but was inaccessible through MQTT. The web server was also missing logic for setting the "<>" (split) option. Should resolve #96