gysmo38/mitsubishi2MQTT #106: Temperature set in Home Assistant bounces to default value

julrich1, 2020-09-27:
I'm hoping there is a simple fix or a way to debug this issue that I'm having with my new install of mitsubishi2MQTT.
Controlling the mini-split via the web interface seems to work just fine, but in home assistant, the temperature is sent, the state is returned correctly, and then temperature is reset to a default value of 50.

Here is the temperature being set on the temp topic:
[image]

Here is the fake state being sent to HA:
[image]

And here is the update with the actual state shortly after, reset to 50:
[image]

Any suggestions on what I might take a look at?

julrich1, 2020-09-27:
A side note but may be related, I'm seeing MQTT timeouts pretty consistently in the HA Mosquitto logs:
`1601167425: Client HVAC_62e0c2 has exceeded timeout, disconnecting.`

julrich1, 2020-09-27:
After a bunch of testing it appears this is related to using Fahrenheit units. Celsius seems to work and set the temperature correctly, but Fahrenheit will not hold the temperature setting when using Home Assistant.

julrich1, 2020-09-27:
These three lines seem to have caused the issue when using Fahrenheit. Commenting them out fixes the issue: https://github.com/gysmo38/mitsubishi2MQTT/pull/101/files#diff-ef77a1dd89510ec969cf6d63c05774a4R1418-R1420

I can debug more later but hopefully someone with more knowledge of the codebase can take a closer look.

melyux, 2020-09-29:
Fixed in #105

julrich1, 2020-09-29:
Thanks! I will test out the new changes in the next few days.

melyux, 2020-09-29:
Yep, just needs the PR situation sorted out first

closed 2021-02-24 by gysmo38