gysmo38/mitsubishi2MQTT #111: state feedback value of null

smartboart, 2020-10-11:
Hello, after update to latest version, i have now this feedback while heating see string below...
mode null and action null, which should show heat and heating..

{"roomTemperature":23.5,"temperature":25,"fan":"1","vane":"1","wideVane":"|","mode":null,"action":null,"compressorFrequency":0}

melyux, 2020-10-11:
I was getting this issue when testing the code (it seemed the mode and action , but I thought I fixed it by making sure the `hpGetMode` and `hpGetAction` functions were returning Strings, though there's still a `.c_str()` value that was there before. I haven't had the problem since then, but this looks like it's still happening.

Maybe someone more versed in C and Arduino strings can take a look at those methods and see if I did something wrong. @gysmo38 @mamrai1

melyux, 2020-10-16:
I've put in PR #102 that stops converting the String into a C string, but I'm not sure if that'll be enough to fix this since I can't reproduce it anymore. @smartboart

smartboart, 2020-10-16:
Hello melyux, 
thanks for your  support.
i have implemented your changes into the projekt on my pc and created a bin file to load the new firmware. I will test if it will help and let you know..

smartboart, 2020-10-16:
Ok, the feedback now is much better. But there is still another problem which worked before...
the mode feedback and control for fan only (idle ) over  mqqt  with mode = FAN is not working so ther is also no feedback...

should look like this... I have 2 units so it is easy for me to compare...
This feedback below which should look like this is not working, even the fan mode cannot be switched on over mqtt while writing the mode with FAN..nothing happens....If i switch on the fan mode via the web gui it is going tofan mode, but the mqtt feedback in this case is: {"roomTemperature":20.5,"temperature":24,"fan":"1","vane":"3","wideVane":"|","action":"off","mode":"off"}

and should look like this:
{"roomTemperature":20.5,"temperature":24,"fan":"1","vane":"3","wideVane":"|","action":"idle","mode":"fan_only"}

brackw, 2020-10-19:
The issue persists with hpGetMode (and similarly hpGetAction) returning null after a ~6 times of being called.

I used mqtt debug to publish rootInfo["mode"] from within hpStatusChanged to see it correctly output 'heat' after reboot before becoming null, but a separate debug statement within hpGetMode function would still publish 'heat'.

It probably has something to do with arduino strings, but I am more hardware than software so if there is something inherently wrong here I am not seeing it.

melyux, 2020-10-20:
@smartboart So you’re saying setting the mode to fan over MQTT `mode/set` is not actually, physically turning the fan on? Or is it just not reporting that it’s on?

@brackw Is that with my PR? I don’t see anything obviously wrong either. I’m not getting the null thing when in cooling mode, strangely. There’s something wrong happening with the strings (or maybe ArduinoJson’s JSON document size allocation?), but I haven’t been able to figure it out either

smartboart, 2020-10-20:
Yes it is not turning physically the fan on over mqtt... And if I turn it on over web gui it is physically on,but it is not reporting it is on over mqtt eather.. .

smartboart, 2020-10-20:
If you want I can upload the project or the bin file I am using without any trouble to compare.. I am using the code from August this year and just changed the things regarding the wifi reconnect and left the other changees away....

brackw, 2020-10-20:
@melyux Yes, this was with your PR code implemented into my local copy.

On a hunch I added:

rootInfo.clear();

Just before new settings are written into rootInfo.

I'm pretty sure this is not the right way to fix issues with buffer issues (if thats even what is happening), but I flashed this change to 4 heatpumps, one is running in cool and one is set to heat (but not running due to current cool request) and all units have not returned null yet.

I may be just masking the issue rather than fixing it.

Unfortunately I cannot replicate the issue @smartboart is having. I can set to fan from mqtt/HA without issue.

smartboart, 2020-10-21:
OK.. If nobody else have this issue, do not waste your time with that. .. I have a solution for me which is working fine... . Thanks for your support in this matter.. .

brackw, 2020-10-21:
@lesleypersyn If you want to try adding rootInfo.clear(); at line 1296 within hpStatusChanged(heatpumpStatus currentStatus) to clear the JsonDocument before filling it with new status info to see if it appears to resolve it for you as well. 

I wouldn't say this is the fix since I don't know the exact root cause and why @melyux isn't seeing it occur. Or if it causes any unforeseen issues.

lesleypersyn, 2020-10-21:
@brackw Add rootInfo.clear(); at line 1296 with latest master, compiled, uploaded.
previous result: {"roomTemperature":21.5,"temperature":18,"fan":"AUTO","vane":"SWING","wideVane":"SWING","mode":null,"action":null,"compressorFrequency":0}
new result:
{"roomTemperature":21.5,"temperature":18,"fan":"AUTO","vane":"SWING","wideVane":"SWING","mode":"off","action":"off","compressorFrequency":0}

This solves the "null" string issue.
I hope this helps the further development.

Stendec1, 2020-10-22:
Hi,

I too had this problem and tried several things to fix it but to no avail. Then I noticed in an old version that the rootinfo buffer was initialized right before populating it in hpStatusChanged. So I added the following lines right before filling-up the rootinfo members; at line 1299:

    const size_t bufferSizeInfo = JSON_OBJECT_SIZE(8);
    StaticJsonDocument<bufferSizeInfo> rootInfo;

Not sure this is the way to go but it certainly fixed the problem for me.

Stendec1, 2020-11-04:
@brackw, @lesleypersyn In the description of Version 6 / API JsonDocument I read the following:

_"I strongly recommend against using a global JsonDocument because you would be troubled by the monotonic allocator, and would make inefficient use of your RAM.
On the contrary, I recommend declaring a short-lived JsonDocument that you use only in your serialization functions."_

Looks like it is defined globally at line 80:
//Local state
StaticJsonDocument<JSON_OBJECT_SIZE(12)> rootInfo;

Using rootInfo.clear() might be a good alternative to declaring a short-lived JsonDocument I guess.

lesleypersyn, 2020-11-04:
@Stendec1 correct, I'm checking this also in depth.
There seems multiple issue's with this topic. I'm working on this...
If you use serializeJson after filling in the data the rootInfo is cleared automatically. But in this case it isn't.
Why it's defined here again with other memory pool sizes is not clear for me. The rootInfo is expanded from the first editions and not adjusted correctly... (i think) See below:
https://github.com/gysmo38/mitsubishi2MQTT/blob/b0eabf4f2f370e33366f01ee73569f04d86ab500/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L1223-L1224
and
https://github.com/gysmo38/mitsubishi2MQTT/blob/b0eabf4f2f370e33366f01ee73569f04d86ab500/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L1326-L1327
JSON_OBJECT_SIZE should be 8 when verifying with [https://arduinojson.org/v6/assistant/](url) The 6 above generates probably a problem!

I'll do some more testing/optimizing...

Stendec1, 2020-11-04:
@lesleypersyn I think I know why it is defined there with another memory pool size. If you don't you get following result: 
[image]

where it should be:
[image]

mamrai1, 2020-11-04:
Sorry I haven't had time to check... For sure the problem lies on latest optimizations from @melyux..... I have to update my code and see.. Maybe somewhere settings and state get mixed?

lesleypersyn, 2020-11-05:
Spend too much time on this. Replaced many parts but same issue.
Found this one: [https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/](url)
The cause is due the change of @melyux to change the location of StaticJsonDocument (to global/local). This is needed due functionality. So solution 2 is the correct solution to handle the function.

Adding rootInfo.clear(); between https://github.com/gysmo38/mitsubishi2MQTT/blob/b0eabf4f2f370e33366f01ee73569f04d86ab500/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L1295-L1297 like @brackw told is a valid solution. ok?

mamrai1, 2020-11-09:
> Spend too much time on this. Replaced many parts but same issue.
> Found this one: [https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/](url)
> The cause is due the change of @melyux to [...]

Pull request and call it a day?

mamrai1, 2020-11-09:
after playing around it seems the solution is not working...at least for me...it throws null values after a couple of hours.

mamrai1, 2020-11-09:
@lesleypersyn Do you use latest swicago library?

mamrai1, 2020-11-10:
@melyux  can you please check your heatpump library you use with latest changes? 
there is a change on latest version of swicago library regarding externalupdates and autoupdate that cause problems

lesleypersyn, 2020-11-10:
@mamrai1, i've seen some questions. In my case i use Heatpump v 1.0.0 (see platformio.ini)

melyux, 2020-11-11:
@mamrai1 I've been using the HeatPump library from April 29, 2020. I haven't updated it to include the 7 commits since then. It'd be weird if that were causing this problem cuz I had this problem before too until expanding the JsonDocument size during dev.

@Stendec1 Good catch, I was using this simply to store the local state but didn't know about that catch.

An alternative solution could be to store the local state in a different way, and transfer the information to an ad hoc JsonDocument whenever we need to serialize, like the documentation suggests. I had found a global JsonDocument to be the simplest and most efficient way to keep state, but obviously it's causing some unforeseen problems.

lesleypersyn, 2020-11-11:
@melyux Checked [https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/](https://arduinojson.org/v6/faq/i-found-a-memory-leak-in-the-library/) already? It's a solution. (solution 2) Not a memory leak.

Stendec1, 2020-11-11:
@lesleypersyn, @melyux, @mamrai1, 
Looks like rootInfo.clear() is the easiest way to go. I had no issues since adding it 20 days ago.

smartboart, 2020-11-11:
Mee too...

lesleypersyn, 2020-12-02:
Can someone check what went wrong with this topic? I published a time ago a PR. The changes seems like they are done but the PR is still open (and now in conflict)
If i check git blame then i can see it's changed due the PR #118 strange? or not? see below;
[https://github.com/gysmo38/mitsubishi2MQTT/blob/7875966340c839cbc110fc1372930e7eb0000caa/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L1223](https://github.com/gysmo38/mitsubishi2MQTT/blob/7875966340c839cbc110fc1372930e7eb0000caa/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L1223)

[https://github.com/gysmo38/mitsubishi2MQTT/blame/7875966340c839cbc110fc1372930e7eb0000caa/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L1223](https://github.com/gysmo38/mitsubishi2MQTT/blame/7875966340c839cbc110fc1372930e7eb0000caa/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L1223)
Thanks for checking.

melyux, 2020-12-02:
From a cursory look, it looks like @wabarkley applied your PR changes on top of his own branch for his own (unrelated) PR #119, which was merged. So I think your changes are in through him now.

wabarkley, 2020-12-02:
> From a cursory look, it looks like @wabarkley applied your PR changes on top of his own branch for his own (unrelated) PR #119, which was merged. So I think your changes are in through him now.

Yeah, one of these days I'll figure out how to do proper configuration management.

lesleypersyn, 2020-12-02:
> > From a cursory look, it looks like @wabarkley applied your PR changes on top of his own branch for his own (unrelated) PR #119, which was merged. So I think your changes are in through him now.
> 
> Yeah, one of these days I'll figure out how to do proper configuration management.

Complex and intuitive ... (still learning every day...)

gysmo38, 2022-01-16:
old issue. Open a new one if still have the issue

closed 2022-01-16 by gysmo38