gysmo38/mitsubishi2MQTT #116: Change via HVAC remote get overwritten.

Stendec1, 2020-11-04:
When changing a setting via the HVAC remote, the value is published ok via ha_settings_topic, ha_state_topic and HA but afterwards overwritten by the old value.

brackw, 2020-11-04:
By default SwiCago's library doesn't let remote updates propagate and reverts them back (by design he did not want remotes to override automation) but he did code in the option for it, it's just not brought out in mitsubishi2MQTT.

` hp.enableExternalUpdate();`  can be called to let remote commands stick.

You must have updated SwiCago's library to rev#164 or higher, as there was a bug in earlier versions that caused it not to work.

I tossed the command into my local code and it works as expected.

Someone should actually add it as a user defined option via the web interface like debug or useFahrenheit and supportHeatMode.

Stendec1, 2020-11-04:
@brackw thanks for the quick answer. I'll have a look at this.

Stendec1, 2020-11-05:
@brackw I added this to the setup() section and it did the trick! The main reason I want to have the "remote" updates stick is that I want the iSee sensor to control the absence detection. Adding it as a user defined option via the web interface might indeed be a welcome option.

Rolf-M, 2021-04-13:
@Stendec1 could you add the exact change you did (the exact file and lines). I'll need to control the AC with the normal remote as well.
Maybe then it'll be easier for someone to add it to the webinterface. This would be a cool feature. The simple infrared remote is a main part of the "woman acceptance factor" for me :-)

Stendec1, 2021-04-13:
@Rolf-M, 

Just put "initCaptivePortal();" near the end of the setup section after the "dnsServer.start..." statement. Somewhere around line 197: 

   ...
   else {
    dnsServer.start(DNS_PORT, "*", apIP);
     _**initCaptivePortal();**_
  }
  initOTA();

Rolf-M, 2021-04-13:
@Stendec1,
I think you mixed something up.... 
But I've seen, that the hp.allowexternalupdate() call is in the latest version from feb. 24th. so I think this issue may be closed.

Now somebody wanting to have the infrared remote disabled would need to comment this out.

Stendec1, 2021-04-13:
@Rolf-M 

Sorry copied the wrong part before. This should be it:

Just put "hp.enableExternalUpdate();" near the end of the setup() section, near line 193:

    // RSC allow remote and iSensor to update settings
    hp.enableExternalUpdate();
  }
  else {
    dnsServer.start(DNS_PORT, "*", apIP);
    initCaptivePortal();
  }
  initOTA();
}

gysmo38, 2022-11-27:
Old issue. Try new version and open a new one if the issue is still there.

closed 2022-11-27 by gysmo38