gysmo38/mitsubishi2MQTT #124: Error when compiling for ESP32

RomeHein, 2021-01-18:
I have this issue when compiling the project for a d1 mini esp32:
```In function 'void saveUnit(String, String, String, String, String, String)':
mitsubishi2mqtt:287:33: error: 'unit_conf' was not declared in this scope
   File configFile = SPIFFS.open(unit_conf, "w");```
   
  I would have wrote a pull request to address this by adding this line:
```const PROGMEM char* unit_conf = "/unit.json";```
in the config.h file but as I'm not able to make this project work properly with an esp32 i'm not sure this would actually work.
I've also noticed this line: 
```const PROGMEM char* advance_conf = "/advance.json";```
which is not being called anywhere in the code and seems to replace the `unit_conf` variable.
Any idea on what's happening here?

sir-buckyball, 2021-01-19:
I just went through this with an ESP32 (specifically an adafruit Huzzah32) and found quite a few patches were necessary which I did locally.

* needed to pass `Serial1` to the heatpump initialization code (since the default goes to USB on my board)
* needed to call WiFi.setHostname _after_ wifi.begin for it to use the configured name
* needed to add missing variables to config.h (as you noted in this bug)
* needed to comment out `WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);`

My local hacks now work, but I think a significant code refactoring is needed in order to cleanly support multiple boards (assuming each has its own quirks).

mattkerrison, 2021-01-19:
@sir-buckyball can you upload your file so I can see it all? I have a spare esp32 whilst waiting for more D1 minis, I ran into similar problems last night and wouldnt mind getting it going

RomeHein, 2021-01-19:
@sir-buckyball could you make a PR to this repo so that we can see your changes?
Also, can you explain the first and last bullet point modifications?

sir-buckyball, 2021-01-19:
> needed to pass `Serial1` to the heatpump initialization code (since the default goes to USB on my board)

This was specifically changing `hp.connect(&Serial);` to `hp.connect(&Serial1);`.

I'll admit I'm not 100% positive this was needed, because I made a hardware change as well (I had been using a logic-level shifting circuit with no luck, and switched to using pullup resistors [despite documentation warnings that the board was not 5v tolerant]). https://learn.adafruit.com/adafruit-huzzah32-esp32-feather/pinouts#logic-pins-2816347-7

> needed to comment out `WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE);`

With this line in the code, I found that my board connected to my wifi network, but reported an ip address of 255.255.255.255 (clearly wrong). When I removed the line, the board started connecting to wifi as expected.

re: my source, I can do that, but not right now; later today.

RomeHein, 2021-01-19:
Nice, thanks for the precision. I've tried Serial1 and even Serial2 but wasn't able to get a connection with my units (they are MSZ-AP15). Do you have an idea how I could debug that?

sir-buckyball, 2021-01-20:
I've uploaded code that compiles for the adafruit esp32 feather (huzzah32) to https://github.com/sir-buckyball/mitsubishi2MQTT/tree/huzzah32/src/mitsubishi2mqtt.  The code is not something I consider clean enough for a pull request.

@RomeHein re:Serial1 vs. Serial2, my understanding is that the esp32 is pretty flexible in its hardware serial support. Ultimately you just need to pass a reference to a HardwareSerial object which has been initialized to pins that match your board. Typically, Serial1, Serial2, ... are defined with the board definitions and map to the various pins that have been labeled for serial communication. I recommend you read the datasheet for the board you are using to ensure the mapping is right. If you have a second board or a protocol analyzer, you could try to confirm that the board is outputting data.

RomeHein, 2021-01-20:
no worries, thanks for the precisions. I've tried both Serial1 and 2 (and of course changed the pin connection accordingly) but can't get any packets from the unit.  I've read somewhere that some people had success by using a different baudrate (2000 to be precise). Anyway thanks again, I'll post an update here if I found something relevant to this issue that could make the esp32 work.

adamderuwe, 2021-01-23:
Anyone know if there is a fork with a more active owner? This seems to be where all the activity is happening but the repo owner seems absent.

jbates58, 2021-01-31:
yeah, i would like to know if there is a fork of this elsewhere?

i am having issues getting it to compile to my WemosD1, but fixing the issues is WELL outside the scope of my knowledge.
I have opened a github issue on it back on Dec 29 2020. Github issue #123 but it has had no responses as of yet.

Jason

gysmo38, 2022-11-27:
Old issue. Try new version and open a new one if the issue is still there.

closed 2022-11-27 by gysmo38