gysmo38/mitsubishi2MQTT #126: Kinda bricked D1 mini?

AkdM, 2021-02-03:
Hey there.

So I've managed to get everything to work, but I noticed that now I just can't flash to something else. Because the USB way couldn't work, I've used the OTA fw upgrade from the web UI, and it apparently worked (to a basic LED blinking). But now that I've done this, it just won't get any info or flash.

Here are the commands I've tried:
`esptool.py -p /dev/cu.usbserial-1420 read_mac`
`esptool.py -p /dev/cu.usbserial-1420 chip_id`
etc.

The only output I have is the following:

```
> esptool.py -p /dev/cu.usbserial-1420 read_mac
esptool.py v3.0
Serial port /dev/cu.usbserial-1420
Connecting........_____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to Espressif device: Timed out waiting for packet header
```

The `Connecting` dots run for like 10-15 seconds then fails with the above error. In the same time, the LED from the chip is blinking about every 2 seconds.

I've of course tried multiple USB cables, multiple ports, tried on another computer and everything, grounded D3 (GPIO0) but nothing worked.
I have another D1 Mini which works, and I do everything the same.

Any of you experienced the issue? I'm on macOS by the way.

Thanks!

lesleypersyn, 2021-02-04:
Power off d1, push button d1, keep pushing, connect usb cable (or power) release button, try to flash...

AkdM, 2021-02-04:
@lesleypersyn Nope, still nothing. It only blinks the blue LED when `esptool` is trying to connect.

lesleypersyn, 2021-02-04:
Best way to be sure is to test with another d1....

AkdM, 2021-02-04:
Yep, I have another one and it works well. I don't event push the RST button, I simply connect the USB, and do things.
But I guess something is broken since even ´read_mac' doesn't outputs anything.

lesleypersyn, 2021-02-04:
Something can be broken.always buy 'spares'!

AkdM, 2021-02-04:
You're right. It's a chance they cost less than $3 each!

reibuehl, 2021-02-04:
I had great success with Tasmotizer (https://github.com/tasmota/tasmotizer) in the past flashing ESPs that I couldn't flash with anything else. Export a Blink sketch for the D1 Mini from Arduino IDE and use Tasmotizer to flash that binary to the D1 and see if this sorts the issue out.

AkdM, 2021-02-05:
Thanks for the input, but it doesn't change anything, since even `esptool read_mac` won't output anything 😞

AkdM, 2021-02-11:
I'm closing it since I think it's a faulty unit.

closed 2021-02-11 by AkdM