gysmo38/mitsubishi2MQTT #127: Wrong Network Config

heffneil, 2021-02-04:
Forgive my ignorance but I must have mistyped the SSID when setting up my system.  The unit is unreachable.  Is there a way to get it back to AP mode or to update the module's settings? Thanks!

Neil

lesleypersyn, 2021-02-04:
Fallback to AP mode should work. Search wifi...

heffneil, 2021-02-04:
Super weird.  I unplugged my AP for that network and I could find it.  I reprogrammed it and for whatever reason, I still can't see it getting an IP from DHCP.  I see the unit rebooting (I think) the blue light blinks after some period of time?

heffneil, 2021-02-04:
Maybe my router just stinks at showing DHCP clients - because now I found it.  Sorry for the trouble- thanks!

melyux, 2021-02-07:
Hey @heffneil, in that case make sure to close this issue (only if it's working now!).

closed 2021-02-08 by heffneil