gysmo38/mitsubishi2MQTT #129: Not working with Wemos Mini clone

fastbike, 2021-02-06:
I have been using this program with 2 heat pumps since last year, following the excellent instructions provided by Chris Davis.
My hardware is IZOKEE Development Board for ESP8266 ESP-12F.

I am helping a friend add automation to his heat pumps but am getting stuck when uploading the most recent version.  The Arduino IDE shows the compile/upload completing successfully but the LED on the board flashes at about 3 Hz (indicating an error ?).  The wifi access point cannot been seen.

If I compile and upload version mitsubishi2MQTT-0.4.7.1 everything works fine.  I can connect to the wifi access point and configure the system.

What changes should I be looking for - the core library of one of the included libraries ?

theonetoday, 2021-07-13:
I am having the same issue. Upload completes without issue, but the light flashes consistently and no WiFi access point is seen. I have this specific model board from Amazon: ACEIRMC ESP8266 ESP-12 ESP-12F NodeMcu Mini D1 Module WeMos Lua 4M Bytes WLAN WiFi Internet Development Board Base on ESP8266 ESP-12F for Arduino,Compatible with WeMos D1 Mini

gysmo38, 2022-11-27:
Old issue. Try new version and open a new one if the issue is still there.

closed 2022-11-27 by gysmo38