gysmo38/mitsubishi2MQTT #133: settings feedback sometimes values with no sense

smartboart, 2021-02-27:
Hello, when i for example change settings vane state or any other setting with mqtt, sometimes the feedback in the settings string looks like this...
Fresh compiled today from master branch 27.02.2021

{"temperature":25,"fan":"3","vane":"SWING","wideVane":"|","mode":"���?X !@��5"}

smartboart, 2021-02-28:
Ok, i have nod widefane in my units and compressor frequency is also not working, so i have deleted this parts in the code.
Since that ia have no problems with this feedback anymore...

The only issue with my solution is that the feedback string takes sometimes a little time to complete after setting changes by mqtt.
in this case it just shows this  {"roomTemperature":27,"temperature":26,"fan":"1","vane":"1","mode":"off"}
and after a while it showsthe complete string.
{"roomTemperature":27,"temperature":26,"fan":"1","vane":"1","mode":"off","action":"off"}

But no stupid signs anymore.

Why you are not making these options widefan and compressorfrequency as a part of the config?

Best regards...

smartboart, 2021-03-01:
ok happened again... in Juli last year the sketch was stable...since that every change made it worse...
I am sad that i donot have the scetch from juli anymore...here all the branches a mixed togehther and still the same version...

mcouture87, 2021-03-01:
I too see the "garbage" state string issue.    It will resolve itself, but for me, the webpage or MQTT will report the unit as 79* until another polling cycle.

mcouture87, 2021-03-01:
Here is an instance where I get garbage data:

{"temperature":73,"fan":"4","vane":"","wideVane":"|","mode":"���?E @���?"}

smartboart, 2021-03-01:
Yes this I really a problem if just work with mqtt... I work with this string by creating separate states for each info with Javascript... This states I use in iobroker as trigger or feedback in vis... This makes no fun. . I am lucky that I found my old binary from June which works perfect.. Only new feature I have worked in was the wlan reconnect which solved the only issue I had that time..

2920, 2021-07-07:
What mode is the unit in when this happens?

mamrai1, 2021-07-28:
yes, i can confirm the garbage data with 'mode' when changing the hvac temperature 
Can't figure out the problem..

gysmo38, 2022-11-27:
Old issue. Try new version and open a new one if the issue is still there.

closed 2022-11-27 by gysmo38