gysmo38/mitsubishi2MQTT #136: Updating via the web console

mattkerrison, 2021-04-27:
I want to update the version running on my ESP8266's without pulling them out, am I right in thinking all I need to do is compile the latest code and then upload the complied code via the web console to update?

produdegr, 2021-04-27:
Once your ESP8266 is on line it should appear under Main Menu -> Tools -> Port in your Arduino IDE.
Select the required device, then press upload. The sketch will be compiled and uploaded to your ESP8266 via OTA.
However if your mitsubishi2MQTT version is older and does not have OTA then I am afraid you must pull them out.

gysmo38, 2022-11-27:
Old issue. Try new version and open a new one if the issue is still there.

closed 2022-11-27 by gysmo38