gysmo38/mitsubishi2MQTT #139: compilation issue

Enjoy-Combi, 2021-05-17:
Hi,

Compilation was successful and after arduino IDE update I get errors.
Hereafter some details: 

Arduino : 1.8.15 (Mac OS X), Carte : "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled (new aborts on oom), Disabled, All SSL ciphers (most compatible), 32KB cache + 32KB IRAM (balanced), Use pgm_read macros for IRAM/PROGMEM, 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

/Users/fredericdeghetto/Downloads/mitsubishi2MQTT-master/src/mitsubishi2mqtt/mitsubishi2mqtt.ino: In function 'void saveUnit(String, String, String, String, String, String)':
mitsubishi2mqtt:272:19: error: invalid conversion from 'char' to 'const char*' [-fpermissive]
  272 |   if (tempUnit == '\0') tempUnit = "cel";
      |                   ^~~~
      |                   |
      |                   char
In file included from /Users/fredericdeghetto/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/Arduino.h:286,
                 from sketch/mitsubishi2mqtt.ino.cpp:1:
/Users/fredericdeghetto/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/WString.h:148:38: note:   initializing argument 1 of 'bool String::operator==(const char*) const'
  148 |         bool operator ==(const char *cstr) const {
      |                          ~~~~~~~~~~~~^~~~
mitsubishi2mqtt:275:18: error: invalid conversion from 'char' to 'const char*' [-fpermissive]
  275 |   if (minTemp == '\0') minTemp = 16;
      |                  ^~~~
      |                  |
      |                  char
In file included from /Users/fredericdeghetto/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/Arduino.h:286,
                 from sketch/mitsubishi2mqtt.ino.cpp:1:
/Users/fredericdeghetto/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/WString.h:148:38: note:   initializing argument 1 of 'bool String::operator==(const char*) const'
  148 |         bool operator ==(const char *cstr) const {
      |                          ~~~~~~~~~~~~^~~~
mitsubishi2mqtt:278:18: error: invalid conversion from 'char' to 'const char*' [-fpermissive]
  278 |   if (maxTemp == '\0') maxTemp = 31;
      |                  ^~~~
      |                  |
      |                  char
In file included from /Users/fredericdeghetto/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/Arduino.h:286,
                 from sketch/mitsubishi2mqtt.ino.cpp:1:
/Users/fredericdeghetto/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/WString.h:148:38: note:   initializing argument 1 of 'bool String::operator==(const char*) const'
  148 |         bool operator ==(const char *cstr) const {
      |                          ~~~~~~~~~~~~^~~~
mitsubishi2mqtt:281:19: error: invalid conversion from 'char' to 'const char*' [-fpermissive]
  281 |   if (tempStep == '\0') tempStep = 1;
      |                   ^~~~
      |                   |
      |                   char
In file included from /Users/fredericdeghetto/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/Arduino.h:286,
                 from sketch/mitsubishi2mqtt.ino.cpp:1:
/Users/fredericdeghetto/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/WString.h:148:38: note:   initializing argument 1 of 'bool String::operator==(const char*) const'
  148 |         bool operator ==(const char *cstr) const {
      |                          ~~~~~~~~~~~~^~~~
mitsubishi2mqtt:284:22: error: invalid conversion from 'char' to 'const char*' [-fpermissive]
  284 |   if (supportMode == '\0') supportMode = "all";
      |                      ^~~~
      |                      |
      |                      char
In file included from /Users/fredericdeghetto/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/Arduino.h:286,
                 from sketch/mitsubishi2mqtt.ino.cpp:1:
/Users/fredericdeghetto/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/WString.h:148:38: note:   initializing argument 1 of 'bool String::operator==(const char*) const'
  148 |         bool operator ==(const char *cstr) const {
      |                          ~~~~~~~~~~~~^~~~
mitsubishi2mqtt:287:24: error: invalid conversion from 'char' to 'const char*' [-fpermissive]
  287 |   if (loginPassword == '\0') loginPassword = "";
      |                        ^~~~
      |                        |
      |                        char
In file included from /Users/fredericdeghetto/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/Arduino.h:286,
                 from sketch/mitsubishi2mqtt.ino.cpp:1:
/Users/fredericdeghetto/Library/Arduino15/packages/esp8266/hardware/esp8266/3.0.0/cores/esp8266/WString.h:148:38: note:   initializing argument 1 of 'bool String::operator==(const char*) const'
  148 |         bool operator ==(const char *cstr) const {
      |                          ~~~~~~~~~~~~^~~~
exit status 1
invalid conversion from 'char' to 'const char*' [-fpermissive]

Is there something wrong in my config ?
Many thanks,
Frederic

Enjoy-Combi, 2021-05-17:
Apparently issues comes in case of esp8266 Community >=3.0.0. With version 2.7.4 it goes well.

closed 2021-05-18 by Enjoy-Combi

akanegally, 2021-09-20:
Hello, I got the same error.
Should I downgrade my esp8266 or correct the compilation errors ?

PS ; I have some trouble to download old version of esp8266 using the board manager.
2.7.4 or 2.7.3 can't be installed because of some download errors : 
i686-w64-mingw32.mksiffs download failed