gysmo38/mitsubishi2MQTT #140: Ducted unit model option

jLynx, 2021-06-02:
Hi it would be awesome to add another model for "ducted units" as they dont have Vanes or Swing.

I have a PEAD-M50JAA ducted unit, so this would then be perfect!

gysmo38, 2022-11-27:
@jLynx You want a unit without Vanes and Swing? Does it work well with Vanes and Swing?

jLynx, 2022-11-28:
My unit does not have Vanes or Swing.

Sprinterfreak, 2022-12-15:
@jLynx Does Your unit work properly and just ignores the vane settings or does the presence of the vane settings some malfunction?

jLynx, 2022-12-15:
Works perfectly fine with no issies

jLynx, 2022-12-15:
Works perfectly fine with no issues at all. It's just to clean up the UI

prashker, 2023-04-10:
Is there a "tell" when unit is ducted or is this more a "user should mark this as ducted" thing? From the perspective of Mitsubishi2MQTT (and probably the upstream library) it seems it believes I have some vane settings.

[image]

```
{
  "temperature": 19.5,
  "fan": "AUTO",
  "vane": "5",
  "wideVane": "<<",
  "mode": "heat"
}
```

closed 2024-08-25 by gysmo38