gysmo38/mitsubishi2MQTT #141: [homeassistant.components.mqtt.climate] temperature parse error

shinyone, 2021-06-03:
Hi, I'm using this module with home assistant integration.  However, I'm not sure where the problem is coming from.

If any of the units is set to the lower temp (16 degrees C) then the following error is shown in the logs.

`ERROR (MainThread) [homeassistant.components.mqtt.climate] Could not parse temperature from `

Here the required temp is set to 17.  And the slider is in position "1"
[image]

I can move the slider via the HA dashboard to 16 (position "0"), but using the remote causes the value not to be unavailable.

The workaround is to manualy move the slider (or not use the lowest temp).    

But where / what is the problem.

adipierro, 2021-06-10:
I have experienced the same. The reason is Home Assistant assuming temperature will be higher than min_value, which is 16° (and lower than max_value, which is 31°) and if it equals, it fails to parse MQTT message. Sadly, don’t know if it is expected behavior or a bug. We probably should ask Home Assistant developers…

shinyone, 2021-06-11:
OK, Thanks for your comment, I will close this bug here then.

closed 2021-06-11 by shinyone