gysmo38/mitsubishi2MQTT #142: Unable to switch heatpump on via MQTT

reibuehl, 2021-06-05:
If "ON" is send to the power/set MQTT topic, the unit is not switched on. 

I think there should be a corresponding else or else if section to the if clause in https://github.com/gysmo38/mitsubishi2MQTT/blob/2b09164a9aca409cfaa18bf70d4d69cce068ecb2/src/mitsubishi2mqtt/mitsubishi2mqtt.ino#L1367

lesleypersyn, 2021-06-05:
If you assign a mode then it will turn ON. That's the logic.

reibuehl, 2021-06-05:
Agree, but why not support just switching it on in the last used mode as that one is already known by the unit?

reibuehl, 2021-06-05:
Changing the code to 

`    if (modeUpper == "OFF") {
      hp.setPowerSetting("OFF");
      hp.update();
    } else if (modeUpper == "ON") {
      hp.setPowerSetting("ON");
      hp.update();
    }
`
seems to work fine on my units. That way a simple ON/OFF logic can be implemented for MQTT similar to what the ON/OFF selection box does in the HTML interface.

closed 2022-01-16 by gysmo38