gysmo38/mitsubishi2MQTT #149: Device self-resets when connected to the heat pump

ChipwizBen, 2021-07-19:
I'm using a Wemos D1 Mini v2 and I have an issue. After flashing either the latest version from master or the tagged release from January 2020, the device creates its own access point and I am able to connect if the device is EITHER powered by USB or by the heat pump. Once I fill in the details of the existing WiFi network, things start to go awry.

If the device is connected via USB and not to the pump, it correctly connects to the WiFi network and has a reliable connection (steady ping, can access web UI). If it's powered by the heat pump, the device is steady when it is acting as its own AP. After putting in the details of the existing WiFi network, it becomes unstable and has an intermittent connection (ping, status web UI often not working) and seems to self-reset often (based on the LED). I wonder if it's trying to read from the pump and seg faults or something and it continuously resets. Any ideas?

2920, 2021-07-21:
You probably have bad reception and its getting kicked off the network and defaulting to AP. Get the D1 pro and an external antenna.

asonkel, 2021-07-21:
Thanks for this info. I will try to improve the Wifi Signal. But i don‘t understand the reason why the unit is switching on if it is reconnecting to the AP.

2920, 2021-07-21:
Did you use the remote? You get funny business if you use the remote. When using automation you are not supposed to use the remote. It cannot track states and all weird stuff happens.

ChipwizBen, 2021-07-21:
Hey @asonkel, you're in the wrong thread. You should be here: https://github.com/gysmo38/mitsubishi2MQTT/issues/148

@2920, the AP is literally underneath the unit, and with the ESP8266 dangling by the connection it's about 160cm to the AP. I can connect to the ESP8266 when it's in AP mode from the other room, so I think the antenna is good.

2920, 2021-07-21:
Both releases you chose have different behavior. I'd suggest picking one and updating the bug report.

Prior to https://github.com/gysmo38/mitsubishi2MQTT/commit/25fff4077ca44181dc148a7324af406dd1081f38 if it drops from wifi or it cannot connect, it will reboot into AP mode after 10 minutes IIRC. This change makes it so it reboots every 5 mins if it cannot connect to wifi.

ChipwizBen, 2021-07-21:
That's the thing, it connects to WiFi fine, I can see it receive an address from the DHCP server and it will respond to ping. Shortly after that, presumably as it tries to read data from the unit, it resets itself.

If I power it via USB with it not connected to the unit, the WiFi connection is very stable. It's as though it's receiving data from the unit that it doesn't like and it's crashing or self-resetting.

asonkel, 2021-07-21:
> Did you use the remote? You get funny business if you use the remote. When using automation you are not supposed to use the remote. It cannot track states and all weird stuff happens.

I have a remote but i prefer to do this via mqtt as soon the units respond. But at general the remote should also work cause in the sw „external updates“ are activated. Or whats the problem if the remote is also used?

2920, 2021-07-21:
> That's the thing, it connects to WiFi fine, I can see it receive an address from the DHCP server and it will respond to ping. Shortly after that, presumably as it tries to read data from the unit, [...]

If you are on master, on start up the lights will flash on and off while connecting to wifi and turn solid once wifi connects. In the same function, it will read from the heatpump. I would suspect if it had issues reading then it would reboot in a very short amount of time after the light turns solid blue (in the order of milliseconds).

I would be checking pin outs and for any cold solder joints, is the heatpump capable of providing stable power?

I'm out of ideas, good luck.

mcouture87, 2021-07-21:
I had an issue where my MQTT server crashed and my Wemos had extremely long response times.    If I removed the MQTT config from the Wemo then the response time of that device went back to normal.    Don’t know if that’s something you are seeing….

Thanks,

-Mike

closed 2022-01-16 by gysmo38