gysmo38/mitsubishi2MQTT #151: Seems like this project is dead - here's another option

adamderuwe, 2021-07-29:
The repo owner seems to have abandoned this project.  If you are willing to use ESPHome, this looks to be a good option:

https://github.com/geoffdavis/esphome-mitsubishiheatpump

ostat, 2021-08-02:
It does currently seem that way. There are quite a few people who have been out there making changes on their own forks. I wonder if anyone else would be prepared to take over at least until @gysmo38 returns?

with regards to the esp home solution, are you able to explain the pros and cons of switching?

2920, 2022-01-16:
So is there a process to change ownership or do we collectively just vote and use some other fork?

gysmo38, 2022-01-16:
I was pretty busy on others projects but I will try to me more active on it. I use it without any issue, that is also why I am not very active on it. I merge last PR and will be more reactive for others :)

gysmo38, 2022-01-16:
I will try to follow PR and this project. Open a new one issue if needed :p

closed 2022-01-16 by gysmo38