gysmo38/mitsubishi2MQTT #152: WiFi - cannot connect

shalak, 2021-08-14:
I'm using Wemos D1 mini clone. I flashed it successfuly and I can see the HVAC_XXXX network. However, I'm unable to connect to it. My device disconnects, because it couldn't get the IP address assignment.

Is there something I'm missing here?

shalak, 2021-08-15:
The issue may be solved by manually assigning IP address of `192.168.1.2` with netmask `255.255.255.0`

857GIT, 2021-08-16:
Having this same issue with 2 different Nodemcu boards ESP-12E & F.

I couldnt resolve it with changing the IP address.

I have tried to connect with android phone and ubuntu laptop.

I dont have enough coding skills to troubleshoot this...

shalak, 2021-08-16:
@857GIT what exactly do you mean by "changing the IP address" and what is your specific issue? You're unable to connect to board's WiFi? Or are you unable to connect to `192.168.1.1`?

The better you describe the exact steps, the more I'll be able to help you :)

857GIT, 2021-08-17:
Yes of course, sorry. 

I'm unable to connect to board's wifi. And as you said in the secont post "The issue may be solved by manually assigning IP ......." it didn't help in my case as I changed it to 192.168.1.2 or anything else.

`//Captive portal variables, only used for config page`
`const byte DNS_PORT = 53;`
`IPAddress apIP(192, 168, 1, 2);`
`IPAddress netMsk(255, 255, 255, 0);`

I can succesfully compile and upload the code as well the SPIFFS files...

reibuehl, 2021-08-17:
@857GIT "manually assigning the IP" referred to your PC/Notebook and not to the ESP board. The ESP Board does not have a DHCP server when running in AP mode. Therefor you need to set the IP of the client that you are using to connect to th ESP's AP to an IP address from the IP range of the ESP AP.

857GIT, 2021-08-17:
@reibuehl Ah yes of course. I should have thought that.

ananiadisp, 2021-09-09:
Have you managed to connect to the board's wifi finally?
I see the wifi AP but i cannot connect either and open the configuration page.

Could someone please help and explain the steps in detail in order to assign manually the IP (on win10)?
Sorry for my noob question but i am a bit newbie to this.

857GIT, 2021-09-11:
@ananiadisp 

Follow this guide: [https://www.tp-link.com/us/support/faq/919/](url) to assign IP to your win10 computer. You need to to assign IP address from same subnet as your esp8266 IP. Example win10 IP 192.168.0.99 and esp8266 IP 192.168.0.1. After this you should able to connect to esp8266 IP address by the internet browser.

I also had problems to get esp8266 to connect my router so I had to assign IP address to my esp8266 (when connecting to AP). My router gave IP address offer but they did not communicate. But I believe this is quite rare problem...

iankgithub, 2021-09-12:
I am so thankful to find this code for the heat pump. Again thanks for the authors efforts. most of the posts I have seen are using the wemos mini's.  Can anyone advise if the code is suitable for the nodemcu chips and  If not what changes will be required.
I compiled and uploaded to the node mcu, but I cant connect to it with wifi  and the onboard blue led stays permantently on.

857GIT, 2021-09-12:
@iankgithub 
Yes it will work with nodemcu. I have it. Thou I have made few changes to the code but I think it should work without any changes.

So you cant connect to the board as when it is in AP mode or you cant get the board to connect to your router?

If the first: You need to be in same subnet as the board is. Check my post just above yours.

If the second: Try googling "nodemcu cant connect to wifi" or similar. If any of those tips wont work try to put this code under line 65
`// NETWORK: Static IP details...`
`IPAddress ip(192, 168, 0, 130);` // Change this to match your routers IP subnet
`IPAddress gateway(192, 168, 0, 254);`  // Your router IP
`IPAddress subnet(255, 255, 255, 0);`
`IPAddress dns(192, 168, 0, 111);` //  Your DNS server IP (same as router if you dont have)
and
`WiFi.config(ip, dns, gateway, subnet);` // <------ ADD This code
`WiFi.begin(ap_ssid.c_str(), ap_pwd.c_str());` // <------- above this line (they are somewhere around line 1620)

You can also uncomment `Serial.println` lines (remove //) and use the serial monitor to debug whats going on with the board

iankgithub, 2021-09-12:
thanks for the reply. Will go through your post and try to understand what is happening.
thanks for confirming it should run on nodemcu chip

iankgithub, 2021-09-12:
I will take a look tomorrow as its getting late here & let you know how I go.
Really looking forward to getting this going for the heat pumps

ananiadisp, 2021-09-12:
@857GIT 
Thanks a lot for your help, you are the best!!
Finally i managed to connect to the esp8266 from the browser and make it to connect to my router.

Now i have to find the CN105 (or jst pa 5pin) to test it with my a/c unit. By mistake i bought an jst ph connector and it doesnt fit.

Do you know where i can by online a jst pa connector, as i cannot find them anywhere in local market(greece)  neither on amazon?

857GIT, 2021-09-12:
@ananiadisp I bought mine from ebay: [link](https://www.ebay.com/itm/383306190385) and [link](https://www.ebay.com/itm/164088478693)

Crimping was a bitch so if you can find premade pigtails anywhere you should go with them.
At the time I ordered those (2020) I couldnt find premade anywhere in europe....

iankgithub, 2021-09-13:
> I will take a look tomorrow as its getting late here & let you know how I go.
> Really looking forward to getting this going for the heat pumps

I'm not able to get back to this for a couple of days due to work but will try the additions to the code and try to get it connected to the router. does the line count for the code include comments? or could you give me the code detail before where the additional code goes please. Dont want to stuff it up.

iankgithub, 2021-09-22:
I didnt have any success in in connecting the wifi.
I tried starting again and updated the ide software. Now when I try to compile the code I get errors (beyond my programming knowledge unfortunately) May be the later version of ide is more stringent on coding? They appear to be conversion errors. I have attached a copy of the errors if anyone can give me some guidance.  
[ide output errors.txt](https://github.com/gysmo38/mitsubishi2MQTT/files/7210933/ide.output.errors.txt)

857GIT, 2021-09-22:
@iankgithub wow you have quite a lot compiling errors. I can only help you with the few on top.

You need to change: 
`if (tempUnit == '\0') tempUnit = "cel";`  to  `if (tempUnit.isEmpty()) tempUnit = "cel";`
`   if (minTemp == '\0') minTemp = 16;`  ---->  ` if (minTemp.isEmpty()) minTemp = 16;` 
etc....
etc....

I dont know about the rest compiling errors... I would maybe delete everything and start from beginning.... 
I used Arduino IDE and I only needed to change those what I mentioned....

iankgithub, 2021-09-22:
thank you will try fixing the invalid conversion errors and see what happens

iankgithub, 2021-09-22:
thank you will try fixing the conversion errors & see what happens.
determined to get this resolved so I can control
my air cons.

iankgithub, 2021-09-26:
Hi 857git,
I changed the code per your advise using .isEmpty() etc and the compiler passed that ok thankyou.  I still get all the library errors. So I tried putting the misubish2mqtt code on to a a windows 7 notebook that I use for compiling tasmota projects using ide.
When I compile it on this pc  it gets through all the code and some libraries and then fails on ctags file not found,

_"Using library HeatPump-master at version 1.0.0 in folder: C:\arduino-1.8.2\sketchbook\sensor1_config_may_2020\libraries\HeatPump-master 
exec: "{runtime.tools.ctags.path}/ctags": file does not exist
Error compiling for board NodeMCU 0.9 (ESP-12 Module)."_

I started looking on forums but yet to find a solution
Strange thing is there is a ctags library  in folder arduino-1.8.2\tools-builder \ctags5.8 arduino11\  in it there is a ctags executable. I tried running it with windows command prompt both on its own and using file names  as arguments but it doesnt return any output  results.  As I really dont know what is happening I am stumped as the ide compiler output doesn't say which file it cant find/index.

857GIT, 2021-09-26:
@iankgithub Sorry cant help you with that. Never had problem like that...

iankgithub, 2021-09-26:
thanks for the reply, will keep searching

lesleypersyn, 2021-09-26:
> thanks for the reply, will keep searching

Try vs code with platformio. This wil open a new world... Never looked back to Arduino software.... (In the past, I lost  a lot of time with the Arduino compiler...)

iankgithub, 2021-09-27:
Finally able to compile code and connect via router to esp8266 hvac control.
despite clean ide reinstall, I found deleting ide does not remove all data and libraries.
Careful delete of all data and reinstalled only the referenced library versions (not the latest versions). Able to compile except for one error. Compiler cannot resolve  m2mqtt_version "0.7.3". Found the zip file download from github in config.h had the const declaration commented out. Uncommented and compiled ok. Thanks for your help.

shalak, 2021-09-28:
> Finally able to compile code and connect via router to esp8266 hvac control.

Congratulations! Please consider creating PR with a README.md update, at the very least :)

maxmacstn, 2021-11-08:
I solved the problem that the mobile devices can't obtain the IP address from the AP Mode by:
1. In `initWifi()` function, move `WiFi.softAPConfig(apIP, apIP, netMsk);` to the line before `WiFi.softAP(hostname.c_str());`
2. Compile and it'll work as expected. No need to manually assign the IP Address :)

iankgithub, 2021-11-08:
terrific thanks for the update

gysmo38, 2022-01-16:
Should be fix in last release :)

closed 2022-01-16 by gysmo38