gysmo38/mitsubishi2MQTT #159: HVAC Status ==> DISCONNECTED

PreNRG, 2022-01-12:
Hi! The integration is great, but several times a year the ESP loses connection to HVAC and it breaks my heating. At the same time, the connection to MQTT is fine. It helps to log into the ESP web interface and click the REBOOT button. Recently, ESP shut down in the middle of the night and it was very cold in the morning. Is there a way to reboot the ESP via MQTT or HTTP request? Then I could write an automation that would check the connection status and reboot the ESP. Thank you.

mattwhibley, 2022-01-28:
I'm having the same issue. The only indication via MQTT of the HVAC being disconnected is that the reported temperature stalls at it's last value. Sending a HTTP request to "http://[IP Address]/?REBOOT=" will trigger a reboot. 

I plan to write a quick work around, something to the effect of "if the temperature doesn't change for 30 minutes, send a request to "http://[IP Address]/?REBOOT=".

PreNRG, 2022-01-28:
Hi! Thank you! I try "http://[IP Address]/?reboot" and don try REBOOT.

PreNRG, 2022-01-29:
I'm thinking of doing it this way: raise the temperature by 1 degree every 30 minutes and lower it by 1 degree after a few seconds. If there are no changes in /state, then need to reboot.

mattwhibley, 2022-01-29:
You could do that. For my situation there is enough natural fluctuation that the temperature is rarely constant for 30 minutes. There's also no harm if it reboots from a false positive, the arduino will just go unresponsive for a few seconds.

sean-leach, 2022-04-17:
I had this same issue, on two different boards. Im thinking I may have bought a V3 based board and not a V2.

Does anyone have a link to an inexpensive multi pack of boards that can wire right into 5V and don’t require extra resistors? Ideally an Amazon US link?

mbbush, 2022-05-13:
https://www.amazon.com/gp/product/B081PX9YFV This is what I'm using. 

I have soldered connectors onto a total of 8 boards so far. Six of them seem to be working properly. I have no idea if the issue is with the board, my (poor) soldering skills, or something else.

mbbush, 2022-07-18:
In that PR, rather than implement a way for the device to be restarted by mqtt, I updated it to automatically attempt to reconnect if the heatpump is connected. There was some existing retry logic, but it had some bugs and didn't always work. This should be more stable.

I also added a counter to the status page telling you how many times it's needed to retry. Since I started running this on my heat pumps last week, that number has stayed at zero.

PreNRG, 2022-07-19:
> In that PR, rather than implement a way for the device to be restarted by mqtt, I updated it to automatically attempt to reconnect if the heatpump is connected. There was some existing retry logic, [...]

Great! I solved the problem like this: MQTT updates the data 1-2 times a minute. If the data is not updated for 5 minutes, the automation ping ESP. If the ESP is online, then automation just reboot the board. If ESP offline - it just power loss.

vamega, 2022-08-16:
> https://www.amazon.com/gp/product/B081PX9YFV This is what I'm using, without any resistors. 4 wires directly from CN105 to 5v, G, Tx and Rx.

I just bought a pack of 5 of these, and ran into the issue described on SwiCago/Heatpump, the boards do not work unless the RX trace on the USB side of the board is cut.

I only tried to use this firmware with 4 of those board, but none of them worked without cutting the trace. You seem to have had a 75% success rate. So to anyone else looking at buying those boards YMMV.

Getting original D1 minis, or S2 minis from Wemos would be ideal, but the long shipping time from Aliexpress might make these worth a shot. Cutting the trace isn't hard, just makes it a little less valuable in terms of reuse.

PR #176 (2022-09-09): use exponential backoff for retrying heatpump connection
Fixes #159 

I noticed that there was already code which retried the connection to the heatpump when it was disconnected, but it had a bug where it never reset the number of tries, so if it tried twice on Tuesday, then successfully reconnected, tried twice on Thursday, and successfully reconnected, and tried once on Friday, with the default count of 5 it would then give up until restarted. 

Now it resets the number of tries after each successful connection. It also never completely gives up trying to connect. 

I also updated the logic  [... cut]
closed 2022-09-09 by gysmo38