gysmo38/mitsubishi2MQTT #165: MQTT "state" should be retained so it can be read by HA/etc. immediately

melyux, 2022-02-22:
Since the "availability" topic now shows whether the state is current or not, I think we can make the MQTT state showing the current temperature, operating state, etc. retained so that it can be retrieved and read immediately by Home Assistant or anything else that's parsing the HVAC data. Without this retain right now, for example, Home Assistant throws a bunch of errors on startup until the first recurring state topic is delivered by this program. If we retain, we can also remove the recurring MQTT state publishing and only publish on changes.

Any objections?

mamrai1, 2022-03-16:
Yes, that will also solve the problem with room temperature not sent in every message

JAAS76, 2022-08-18:
+1 for me please,

gysmo38, 2022-11-27:
Please try last version. I change template for HA. It should fiw the 26 error. Not sure for the rest. Open new issue if there is still an issue.

closed 2022-11-27 by gysmo38