gysmo38/mitsubishi2MQTT #168: Wi-Fi Settings not Saved

BreBar07, 2022-03-24:
Hello All,

I am using Wemo D1 Mini Pro, after flashing I get following error;

`mitsubishi2MQTT-master: no headers files (.h) found in ...`
Assume this is ok?

After flashing I get to initial config screen, enter SSID / PSK and reboot. Then after reboot it just defaults back to Initial config state.

Any thoughts? Help?

BreBar07, 2022-03-24:
I manually changed sketch to include SSID / PSK and initially that worked. However after power down config for Wi-Fi / MQTT was lost.

JAAS76, 2022-05-16:
I had this too on two Wemos d1s, I lowered the d1 mini upload speed and it worked for me.

Dreamswielder, 2022-05-28:
Any other way to solve this, or to check what is happening? I loop on the wifi settings and or never conects or never saves... I have tried all upload speeds...

BreBar07, 2022-05-28:
After I manually changed the sketch and flashed the chips it has worked perfectly.

Flashed Initial SSID and PSK, and I have recently rebuilt my network and updated to new SSID and PSK and they are working flawlessly.

Dreamswielder, 2022-05-28:
What did you change in the sketch? Im using Arduino to upload... just in case you are using other program...

gysmo38, 2022-11-27:
Tested with last version, no issue. Please try again.

closed 2022-11-27 by gysmo38