gysmo38/mitsubishi2MQTT #169: Loading Web UI Knocks Offline

Sunoo, 2022-04-05:
MQTT seems to be working, but whenever I try to load the web UI currently it just clocks and I see the availability flip to offline. Sometimes MQTT will come back after a bit, but sometimes not and I need to pop the breaker to get the device to reboot.

I have no idea what’s wrong, but given the behavior it doesn’t seem like a signal strength or hardware issue…

If it matters, I’m running the exact ESP-01 model suggested in the Swicago Heatpump readme.

mcouture87, 2022-08-29:
There is something wrong when MQTT is enabled.   The Web UI and really the TCP stack gets hosed....response time and TCP time-outs happen and makes the devices almost un-usable

gysmo38, 2022-11-27:
Old issue. Please try new version

closed 2022-11-27 by gysmo38