gysmo38/mitsubishi2MQTT #177: How to enable iSave mode (=heating, 10C)?

rhk, 2022-08-05:
Hi!

First - this is an awesome project as all the other projects this builds on (like SwiCago's HeatPump)!! I got this up&running ina very short time - after wanting to do it for years, but not having the guts to connect anything to the expensive HVAC :D 

Oone question/issue:
Main unit: MSZ-FH25VE2
Remote:SG15C
The remote has an 'i save' button, that can be used to set the temp to 10C, when in heating mode.
This is handy in wintertime to keep the house from freezing / dry, but not overheat. 

I attached full MQTT debug log and mitsubishi2MQTT screenshot, where the isave button is pressed  and 10C temperature is shown.
(interesting though, sometimes the log shows temp=0 when this mode is on..)

[heat_10C_log.txt](https://github.com/gysmo38/mitsubishi2MQTT/files/9273644/heat_10C_log.txt)
[image]

-> So my question/issue: Is it possible to somehow enable this mode with mitsubishi2MQTT?

TiceRex, 2022-10-03:
You can find the answer here:
https://github.com/SwiCago/HeatPump/issues/39#issuecomment-284119724
"I think iSave is really just a saved set of settings so I doubt anything will be in the packets". So, there is no special command for iSave, it is just a set of settings.

gysmo38, 2022-11-05:
It'right, iSave have no packet with the unit. I have already give a try few years ago. I think you can simulate iSave with automatisation in your home system.

closed 2022-11-05 by gysmo38