gysmo38/mitsubishi2MQTT #178: Occasionally sends incomplete state.

brackw, 2022-08-20:
Seem to occur generally after fresh reboot, changing the operation mode or temperature results in an incomplete state topic being sent to HA, seems to manifest in HA showing incorrect information until the next complete state topic is sent.

For example:
Recent reboot, received complete state topic:
`{"roomTemperature":71,"temperature":70,"fan":"AUTO","vane":"AUTO","wideVane":"|","mode":"cool","action":"idle","compressorFrequency":19}`
set operation mode to 'dry' and received runt state topic:
`{"temperature":70,"fan":"AUTO","vane":"AUTO","wideVane":"|","mode":"dry","action":"drying"}`
HA now shows room temp as 79 until the next state topic comes in again:
`{"roomTemperature":71,"temperature":70,"fan":"AUTO","vane":"AUTO","wideVane":"|","mode":"dry","action":"idle","compressorFrequency":18}`
Everything back to normal.

I suspect it might be the dummy packet.

brackw, 2022-08-20:
Seems like this would be fixed by the implementation of #165. 

Closing as duplicate.

closed 2022-08-20 by brackw

floatplane, 2023-02-20:
@brackw OOC do you think that this was actually fixed? I'm relatively new to this project (just got everything up and running this weekend) but I'm seeing incomplete state quite frequently - `roomTemperature` is often missing in the state packets, even though I'm pushing the room temperature every 15 seconds, and the web interface on the ESP does reflect the `roomTemperature` value that I'm sending.

floatplane, 2023-02-20:
Should have mentioned that I'm seeing this with latest master (3f1f95abf6a10dc75a99807dce0b676535f57847)