gysmo38/mitsubishi2MQTT #179: Not able to change mode in HA after using latest SwiCago/HeatPump library

arnerek, 2022-09-01:
Hi,

First of all I must say, thanks a lot for the great code. It works really good in my set-up.

I discovered that the new SwiCago/HeatPump library (Aug 16) makes the mode change not working anymore from HA. It works in the HTML pages, but not via MQTT. It updates in HA, but after a minute the old mode is reverted. It seems that the mode change request is not sent to the heatpump. Switching back to the version from April 26 resolves it again. Anyone else with the same experience?

singlecell83, 2022-09-08:
I am finding the same. I only set everything up yesterday for the first time and was looking for a solution. Whatever I set in HA reverts back to what is set in the 8266s web interface after about a minute. That interface will change if you use the remote though. Is there any disadvantage to changing back to the april release?

bigthrilla, 2022-10-11:
Yes, I have that going on as well.  I can shut off/turn on and  change temp and all of that seems to work.  

In my case I can even send the mqtt messages to change mode and that works (both from HA and MQTT Explorer), but if they come from the HA thermostat card I see the state topic change, but the settings topic does not - eventually state topic flips back to align with the settings.  

[image]

@arnerek Thanks for the post, I've been trying to puzzle this for days.  Now to figure out how to switch the version back to the April 26 version.......

juuzonieminen, 2022-10-30:
Did you find solution for this? I have same problem.

bigthrilla, 2022-10-30:
> Did you find solution for this? I have same problem.

Yes, reverted back as @arnerek mentioned above and it's working great.

juuzonieminen, 2022-10-30:
> > Did you find solution for this? I have same problem.
> 
> Yes, reverted back as @arnerek mentioned above and it's working great.

Ok great! Then stupid question: where i can find this april 26 version? Github shows no releases for that library?

bigthrilla, 2022-10-30:
> > > Did you find solution for this? I have same problem.
> > 
> > 
> > Yes, reverted back as @arnerek mentioned above and it's working great.
> 
> Ok great! Then stupid question: where i can find this april 26 version? Github shows no releases for that library?

https://github.com/SwiCago/HeatPump/tree/ca4b8f95d6eff88915d48408a07083a1f16c1d8a

Took me awhile to figure out where github was hiding the commits myself!

gysmo38, 2022-11-27:
Should work with last version.

closed 2022-11-27 by gysmo38