gysmo38/mitsubishi2MQTT #182: HVAC Control portion of Web UI dashboard doesn't seem to work at all (other than temp)

MrChuckDanger, 2022-10-18:
Is it just me, or does the HVAC control portion of the web UI not seem to work except for Set Temp setting? I am unable to set power, mode, fan, vane, or wide vane from the UI - as I monitor the MQTT topics, I do not see my "settings" update. Verified that posts are working fine. Note the following:

When modifying the beginning of the change_states function on line 1202 as follows:
```
heatpumpSettings change_states(heatpumpSettings settings) {
  if (server.hasArg("CONNECT")) {
    hp.connect(&Serial1);
  }
  else {
    bool update = false;
    if (server.hasArg("POWER")) {
      settings.power = server.arg("POWER").c_str();
      if (_debugMode) mqtt_client.publish((ha_debug_topic + "msg").c_str(), (char*)("Power: ")); //debugmsg
      if (_debugMode) mqtt_client.publish((ha_debug_topic + "msg").c_str(), (char*)(server.arg("POWER").c_str())); //debugmsg
      if (_debugMode) mqtt_client.publish((ha_debug_topic + "msg").c_str(), (char*)(settings.power)); //debugmsg
      update = true;
    }
```

I get the following three messages on my "debugmsg" MQTT topic:

```
1. Power:
2. ON
3. �?/
```

Perhaps this is my unfamiliarity with C++, but `settings.power` was set equal to `server.arg("POWER").c_str()`, shouldn't \# 2 and \# 3 show identical output?

Confirming the above:
- If I insert `hp.setPowerSetting(settings.power);` just after the `hp.setSettings(settings);` at the end of the function, behavior is unchanged.
- If I insert `hp.setPowerSetting(server.arg("POWER").c_str());` just after the `hp.setSettings(settings);` at the end of the function, **the power setting is updated appropriately.**

I noticed this as I was able to set the power setting properly by publishing to MQTT, which calls hp.set functions explicitly rather than building an heatpumpsettings object and calling hp.setSettings.

juuzonieminen, 2022-10-30:
I have same problem. Did you find solution?

MrChuckDanger, 2022-10-30:
I have not. I suspect it has something to do with data types and/or assignments given the difference in my example output #2 vs #3, but I am not quite familiar with C enough to understand. Will post back if I find something, but I’m primarily using the MQTT implementation which seems to work without issue.Sent from my iPhone

maxmacstn, 2024-04-29:
I have the same issue. The solution for me was to copy the string before assigning it to the `setting`

```
  if (server.hasArg("CONNECT")) {
    hp.connect(acSerial);
  }
  else {
    bool update = false;
    if (server.hasArg("POWER")) {
      settings.power = strdup(server.arg("POWER").c_str());
      Log.ln(TAG, "Power = " + String(settings.power));
      update = true;
    }
    if (server.hasArg("MODE")) {
      settings.mode = strdup(server.arg("MODE").c_str());
      Log.ln(TAG, "Mode = " + String(settings.mode));
      update = true;
    }
    if (server.hasArg("TEMP")) {
      settings.temperature = convertLocalUnitToCelsius(server.arg("TEMP").toInt(), useFahrenheit);
      Log.ln(TAG, "Temp = " + String(settings.temperature));
      update = true;
    }
    if (server.hasArg("FAN")) {
      settings.fan = strdup(server.arg("FAN").c_str());
      Log.ln(TAG, "Fan = " + String(settings.fan));
      update = true;
    }
    if (server.hasArg("VANE")) {
      settings.vane = strdup(server.arg("VANE").c_str());
      Log.ln(TAG, "Vane = " + String(settings.vane));
      update = true;
    }
    if (server.hasArg("WIDEVANE")) {
      settings.wideVane = strdup(server.arg("WIDEVANE").c_str());
      Log.ln(TAG, "WideVane = " + String(settings.wideVane));
      update = true;
    }
```

ps. Please ignore `Log.ln()` function.

closed 2024-08-25 by gysmo38