gysmo38/mitsubishi2MQTT #184: Today's release seems to break remote temp

disruptivepatternmaterial, 2022-11-12:
[image]

In the above you can see my system is setting remote temp, but in the status it shows as 0.

There seems to be a change around ~1450 that broke this, trying to figure it out now.

disruptivepatternmaterial, 2022-11-12:
if (_debugMode) mqtt_client.publish(ha_debug_topic.c_str(), (char*)("remote temperature set"));
    lastRemote = temperature;
    lastRemoteSet = millis();

at line 1452 is as far as I have been able to see...

gysmo38, 2022-11-12:
Sorry about that bug. I do not user remote temp so i do not test it. I will try to fix it.

gysmo38, 2022-11-12:
I check, now the code is at lien 1438 to 1441.
The only change is that I remove hp.update() because I use hp.enableAutoUpdate(); line 183.
Can you try add hp.update() after line 1440 and test if it resolve the issue ?

disruptivepatternmaterial, 2022-11-12:
Sorry not following you, there is nothing about remote temp around 1438-1441, here is 1400-1466

`        rootInfo["action"] = "idle";
        modeUpper = "AUTO";
      } else if (modeUpper == "HEAT") {
        rootInfo["mode"] = "heat";
        rootInfo["action"] = "heating";
      } else if (modeUpper == "COOL") {
        rootInfo["mode"] = "cool";
        rootInfo["action"] = "cooling";
      } else if (modeUpper == "DRY") {
        rootInfo["mode"] = "dry";
        rootInfo["action"] = "drying";
      } else if (modeUpper == "FAN_ONLY") {
        rootInfo["mode"] = "fan_only";
        rootInfo["action"] = "fan";
        modeUpper = "FAN";
      } else {
        return;
      }
      hpSendLocalState();
      hp.setPowerSetting("ON");
      hp.setModeSetting(modeUpper.c_str());
    }
  }
  else if (strcmp(topic, ha_temp_set_topic.c_str()) == 0) {
    float temperature = strtof(message, NULL);
    float temperature_c = convertLocalUnitToCelsius(temperature, useFahrenheit);
    if (temperature_c < min_temp || temperature_c > max_temp) {
      temperature_c = 23;
      rootInfo["temperature"] = convertCelsiusToLocalUnit(temperature_c, useFahrenheit);
    } else {
      rootInfo["temperature"] = temperature;
    }
    hpSendLocalState();
    hp.setTemperature(temperature_c);
  }
  else if (strcmp(topic, ha_fan_set_topic.c_str()) == 0) {
    rootInfo["fan"] = (String) message;
    hpSendLocalState();
    hp.setFanSpeed(message);
  }
  else if (strcmp(topic, ha_vane_set_topic.c_str()) == 0) {
    rootInfo["vane"] = (String) message;
    hpSendLocalState();
    hp.setVaneSetting(message);
  }
  else if (strcmp(topic, ha_wideVane_set_topic.c_str()) == 0) {
    rootInfo["wideVane"] = (String) message;
    hpSendLocalState();
    hp.setWideVaneSetting(message);
  }
  else if (strcmp(topic, ha_remote_temp_set_topic.c_str()) == 0) {
    float temperature = strtof(message, NULL);
    hp.setRemoteTemperature(convertLocalUnitToCelsius(temperature, useFahrenheit));
    if (_debugMode) mqtt_client.publish(ha_debug_topic.c_str(), (char*)("remote temperature set"));
    lastRemote = temperature;
    lastRemoteSet = millis();
    hp.update();
  }
  else if (strcmp(topic, ha_debug_set_topic.c_str()) == 0) { //if the incoming message is on the heatpump_debug_set_topic topic...
    if (strcmp(message, "on") == 0) {
      _debugMode = true;
      mqtt_client.publish(ha_debug_topic.c_str(), (char*)("Debug mode enabled"));
    } else if (strcmp(message, "off") == 0) {
      _debugMode = false;
      mqtt_client.publish(ha_debug_topic.c_str(), (char *)("Debug mode disabled"));
    }
  }`

gysmo38, 2022-11-12:
I think you do not watch the last version of the file. On the last version, I remove all hp.update()

maggarwal, 2022-11-13:
Hi, latest updated seems to have broken `packetRecv` debug messages. I don't see any `packetRecv` messages in mqtt client anymore. Also, now the sent packets are shown in parsed values and not in hex.

{
  "roomTemperature": 79,
  "temperature": 78,
  "fan": "4",
  "vane": "AUTO",
  "wideVane": "SWING",
  "mode": "heat",
  "action": "heating",
  "compressorFrequency": 0
}

VS 

{"packetSent":"fc 42 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77 "}

gysmo38, 2022-11-16:
@disruptivepatternmaterial  have you any news?
@maggarwal ok I known what happen, i will fix it this week-end. For next time, open a new issue.

mayankst, 2022-11-16:
@gysmo38 

> @disruptivepatternmaterial have you any news? @maggarwal ok I known what happen, i will fix it this week-end. For next time, open a new issue.

Here #186,  thank you

gysmo38, 2022-11-27:
No new from @disruptivepatternmaterial . I close for now because do not able to reproduce the isse.

closed 2022-11-27 by gysmo38