gysmo38/mitsubishi2MQTT #185: Power consumption

cody82, 2022-11-12:
Is it somehow possible to get the current electric power consumption of a MSZ-AP25VGK air conditioner?
I want to use it to calculate the heating power.

tux-, 2022-11-16:
I was wondering the exact same thing. Tried to find information about this myself, but to no luck. I have a MSZ-LN35VGW unit here. So far things seems to work really nice! (Just set it up like 15 minutes ago), only missing power consumption info (if that is even possible?)

gysmo38, 2022-11-16:
There is no power consumption information send by the unit to the module. If you want this type of information, you will need to add external power consumption  plug on the power supply.

closed 2022-11-16 by gysmo38

cody82, 2022-11-16:
Just for information: Even if it is not possible right now with open software, die CN105 connector does provide at least rudimentary power consumption data. Because right now I have the official Mitsubishi WIFI module attached an it provides the used energy to the MelCloud.

tcgerhard, 2025-01-23:
I've been curious if the compressor frequency could be mapped to power consumption.  It would necessarily vary from model to model, but someone with a deep knowledge of the systems might know.