gysmo38/mitsubishi2MQTT #193: ESP32 OTA update in-place not working

Sprinterfreak, 2022-11-28:
If the ESP32 is connected to an HVAC, trying to OTA update a new firmware.bin via WebUI fails somwhere around or after upload stage.

The ESP32 becomes unresponsive and starts to blink its LED periodically as if it is in a boot loop forever. The rhythm looks pretty close to the unit poll interval.
Probably the HVAC connected to Serial0 interrupts the update routine badly.

Luckily after a powercycel everything is back to normal on the previous version.

A non-connected ESP32 updates fine via WebUI.

What did I expect?
In-place update possible while ESP is connected to the HVAC

Sprinterfreak, 2022-12-04:
Seems that enabled debug mode (Setup/Others/Debug: ON) prevents the update to finish.
Turn off debug mode solved this issue so far.

closed 2022-12-04 by Sprinterfreak

giuseppeg88, 2023-01-06:
I'm having the same problem but mine is staying unresponsive.  Is there a way to make the ESP work again without physically unplugging it from the air conditioner? Thanks